victronenergy/gui-v2 #1233: Empty / invalid dbus service displayed on the device list

ReinvdZee, 2024-06-13:
Discovered when testing a genset on a GX with no other devices connected. 
Even though there is a dbus service for `vebus` with a lot of empty paths, GUI-v2 should not display it when it is not valid.
Gui-v1 does not display the invalid device in the device list. 

[image]

This also leads to an empty control card: 
[image]

blammit, 2024-06-14:
@ReinvdZee the Device List assumes that a device is valid if the /DeviceInstance is set, which is incorrect in this case. What would be a good test for validity? How about checking `/DeviceInstance`, `/ProductName` and `/ProductId` are all valid?

ReinvdZee, 2024-06-17:
@blammit Gui-v1 finds a device valid if `/ProductName` or `/CustomName` is set. That is why in this case it works, because both are not set. However, I think V2 should use a more robust test for validity, so I agree with your suggestion:

**A device is valid if `/DeviceInstance`, `/ProductName` and `/ProductId` are all valid and set.** 

Keep in mind that `/CustomName` should be displayed when set, if not, then `/ProductName` is used.

blammit, 2024-06-18:
Sounds good, thanks!

blammit, 2024-06-27:
Discussed that /ProductId should not be mandatory. Just look for /DeviceInstance and /ProductName.

chriadam, 2024-07-05:
Discussed with Rein that the isValid check is currently, wrong.  It is currently:
```
bool BaseDevice::isValid() const
{
	return !m_serviceUid.isEmpty() && !m_productName.isEmpty() && m_deviceInstance >= 0;
}
```

but it should be something like:
```
bool BaseDevice::isValid() const
{
	return !m_serviceUid.isEmpty() && (!m_productName.isEmpty() || !m_customName.isEmpty()) && m_deviceInstance >= 0;
}
```

chriadam, 2024-07-05:
And, in the device list, if there is a custom name set, the custom name should be shown rather than the product name.

blammit, 2024-07-08:
Okay, sounds like this previous conclusion was incorrect:

>  Discussed that /ProductId should not be mandatory. Just look for /DeviceInstance and /ProductName.

blammit, 2024-07-08:
> And, in the device list, if there is a custom name set, the custom name should be shown rather than the product name.

Yep, this is how the name is generated in `components/Device.qml`.

commit aae02e8f (2024-06-18): Do not show invalid devices in the Device List

Invalid devices may still have a valid /DeviceInstance, so tighten the
device validity check to also test for /ProductId and /ProductName.

Any code that uses onValidChanged to update a device model should do
this on the next event loop, to avoid spurious model changes on
initialization.

Fixes #1233
commit cdb3cbf8 (2024-06-26): Do not show invalid devices in the Device List

Invalid devices may still have a valid /DeviceInstance, so tighten the
device validity check to also test for /ProductId and /ProductName.

Do the validity check in BaseDevice so that the test condition is the
same for all devices.

Fixes #1233
commit 6308d2a5 (2024-06-26): Do not show invalid devices in the Device List

Invalid devices may still have a valid /DeviceInstance, so tighten the
device validity check to also test for /ProductId and /ProductName.

Do the validity check in BaseDevice so that the test condition is the
same for all devices.

Fixes #1233
commit c3b81795 (2024-06-27): Do not show invalid devices in the Device List

Invalid devices may still have a valid /DeviceInstance, so tighten the
device validity check to also test for /ProductId and /ProductName.

Any code that uses onValidChanged to update a device model should do
this on the next event loop, to avoid spurious model changes on
initialization.

Fixes #1233
closed 2024-07-10 by blammit