victronenergy/gui-v2 #1234: Chinese translation file is not updating with the latest translations IDs

jpetrell, 2024-06-14:
Reported by the [customer](https://community.victronenergy.com/questions/292058/venus-os-v34026-available-for-testing.html?childToView=294132#answer-294132), some translation IDs like common_words_battery and battery_state_of_health don't have entries on the <TS> file.

[image]
[image]

Looks like Chinese translations have different set of strings compared to the other languages:

```
grep -ric "<translation>" i18n/
i18n/venus-gui-v2_zh_CN.ts:1783
i18n/venus-gui-v2_ar.ts:1772
i18n/venus-gui-v2_cs.ts:1772
i18n/venus-gui-v2_da.ts:1772
i18n/venus-gui-v2_de.ts:1772
i18n/venus-gui-v2_es.ts:1772
i18n/venus-gui-v2_fr.ts:1772
i18n/venus-gui-v2_it.ts:1772
i18n/venus-gui-v2_nl.ts:1772
i18n/venus-gui-v2_pl.ts:1772
i18n/venus-gui-v2_ro.ts:1772
i18n/venus-gui-v2_ru.ts:1772
i18n/venus-gui-v2_sv.ts:1772
i18n/venus-gui-v2_th.ts:1772
i18n/venus-gui-v2_tr.ts:1772
i18n/venus-gui-v2_uk.ts:1771
```

jpetrell, 2024-06-14:
For some reason there are 166 translation IDs missing in the translations, out of which there are 38 genuinely new strings that need translating from the latest new genset, AC charger and modbus settings upstreamed from gui-v1. All those do get translated during compilation, I guess because there is another translation ID with the same Engineering English translation.

jpetrell, 2024-06-14:
Ok apparently this is POEditor bug, it doesn't like identical translation IDs, but also it doesn't like identical Engineering English translations either. Need to add 166 translation comments to make them show as separate translations to the translation pipeline. 

This is bit too easy mistake to make, whenever you introduce new string you need to hunt down all duplicates, and if there are any either declare in CommonWords or add a comment. We can work around it for now, and in the future try to review stuff carefully, but the issue won't fully go away until the toolchain is fixed.

[3:35](https://victrondevelopment.slack.com/archives/D06Q7PBUZ7A/p1718368511986819)
We can work around it, and try to review stuff carefully, but the issue won't go away.

chriadam, 2024-06-18:
I don't understand the bug - does it mean that POEditor doesn't differentiate between two different strings with different translation ids if the engineering english for those strings is the same?  That is ... a bad bug, since it means that some words like "back" or "control" or "switch" which can either be a verb or a noun, for example, cannot be translated properly in POEditor.

Alternatively, if you simply mean that when we added the strings to our codebase, we did not give them unique ids, then that is our bug, we need to provide meaningful and separate translation ids for each string.

jpetrell, 2024-06-19:
>does it mean that POEditor doesn't differentiate between two different strings with different translation ids if the engineering english for those strings is the same?

Yeah, POEditor seems ill-suited to handle Qt translation IDs.

> a bad bug, since it means that some words like "back" or "control" or "switch" which can either be a verb or a noun, for example, cannot be translated properly in POEditor.

You can create separate translations for the same string if you define translation comment, e.g. `//: Use as a verb`.

jpetrell, 2024-06-19:
Started to add translation comments to 166 strings, but gave up after adding dozen or so. Would rather find a proper architectural solution, filed #1256.

Fixed the issue described here "Chinese displaying translation ID instead of the actual translation" in [PR](https://github.com/victronenergy/gui-v2/pull/1255).

PR #1255 (2024-06-19): Fix Chinese translations not updating 
After that Chinese version no longer has translation IDs showing on the GUI. There are still 166 translations missing from POEditor, but those are at least autofilled during build.

Fixes #1234
commit 4d1e6171 (2024-06-19): Fix Chinese translations not updating

Fixes #1234
closed 2024-06-19 by blammit