victronenergy/gui-v2 #1239: Align genset settings UI behavior with similar aspects of inverter/charger control card

blammit, 2024-06-16:
Auto start setting differs from the control cards equivalent in two aspects

- Doesn't show dialog when toggled off
- Isn't disabled when the generator is running

Also use "Manual Start" for button to align with similar text.

blammit, 2024-11-07:
This may already be completed. First task is to verify whether the generator control card behaves in the same way as the generator settings page in regards to starting and stopping the generator. If not, then the behaviors should be aligned.

MikeTrahearn-Qinetic, 2024-11-12:
@ReinvdZee I have confirmed that the confirmation dialog is shown when attempting to turn off the autostart switch in the Control Card, but not when the same is attempted in the settings.
Would you like the confirmation dialog to be shown in the same way for the settings also?

@blammit For reference: the ListSwitch in the PageGensetModel.qml would have to have updateDataOnClick set to false, and the onClick handled as per GeneratorCard.qml where if !checked we ListSwitch.dataItem.setValue(true) to turn it else show the dialog. On dialog accepted, it shall use the ListSwitch.dataItem.setValue(false) to turn it off. Note that the ListSwitch will need an id to make this work.

ReinvdZee, 2024-11-13:
@MikeTrahearn-Qinetic, nice work spotting these differences!

1. I think the confirmation dialog is very useful as we can explain the consequences of turning off autostart. Please implement that for the autostart switch in the settings page as well.
2. In gui-v2, I'm able to turn off autostart from the settings menu and from the control card while the generator is running due to a condition: 
[image]
It has always been possible to disable autostart, even if the generator is running. In This can even be useful when autostart started the generator due to some condition, but the generator must actually be turned off (e.g., to do maintenance). The confirmation dialog (1) is also very useful in this case. We could amend the dialog to saying that disabling autostart will stop the generator if it was running.
3.  `com.victronenergy.generator.startstop1/Enabled` is used specifically for connected (modbus TCP) gensets which require a separate relay to be started and stopped. These gensets can be read out over modbus TCP, but can only be started/stopped with a relay. We recently implemented a feature which allows the user to configure relay 1 as "Connected genset helper relay". The path `com.victronenergy.generator.startstop1/Enabled` will be 1 when the connected genset can be controlled, so either its a regular connected genset, or its a connected genset which requires a helper relay AND the helper relay is configured properly. If this path is not 1, we show a message in the genset's device page: 
[image]
It would be useful to notify the user from the control card as well if the helper relay is not configured. We can also consider not showing the control card, because the genset cannot be controlled in this case. The control card would then pop up when `/Enabled` is 1. @philipptrenz what do you think?

Let me know if you need directions on how to test things.

@mpvader FYI.

philipptrenz, 2024-11-13:
Hi all, 

to 1.: Yes I agree, the popup should be shown in both cases when disabling autostart.

to 2.: Yes, adding that note to the popup text is a nice improvement, here's a proposal:

> Autostart will be disabled and the generator won't automatically start based on the configured conditions.
>If the generator is currently running due to a autostart condition, disabling autostart will also stop it immediately.

to 3.: I would also suggest to *not* show the generator control card if `/Enabled` equals 0.

MikeTrahearn-Qinetic, 2024-11-14:
@blammit would you be able to advise further on the specific implementational details regarding com.victronenergy.generator.startstop1/Enabled and how or when it should be applied?
I'm looking at Generator.qml to potentially make a modification there so that the GeneratorCard can change its visibility within the Repeater in the ControlCardsPage as one possible solution, but I would like to review this with you.

MikeTrahearn-Qinetic, 2024-12-11:
@blammit is there anything else to do on this?

blammit, 2024-12-19:
> @blammit would you be able to advise further on the specific implementational details regarding com.victronenergy.generator.startstop1/Enabled and how or when it should be applied?
I'm looking at Generator.qml to potentially make a modification there so that the GeneratorCard can change its visibility within the Repeater in the ControlCardsPage as one possible solution, but I would like to review this with you.

Sorry about the hold-up, I thought we needed to abstract some of the PageGensetModel code out to find the startstop1 service for a genset, but now I realize we don't need to do that here. From Generator.qml we can just add a property to look up the `/Enabled` value and hide it in the control cards, as you mentioned.

MikeTrahearn-Qinetic, 2024-12-23:
DONE - see PR https://github.com/victronenergy/gui-v2/pull/1812 for a review.

commit b62d33dc (2025-01-06): Generator Control cards should not be displayed when startStop1 disabled

Fixes #1239
commit b4d116ab (2025-01-06): Generator Control cards should not be displayed when startStop1 disabled

Fixes #1239
commit b6aa09b3 (2025-01-08): Generator Control cards should not be displayed when startStop1 disabled

Fixes #1239
commit d8095159 (2025-01-08): Generators should be visible in Control Cards when using mock mode

Fixes #1239
closed 2025-01-08 by MikeTrahearn-Qinetic