victronenergy/gui-v2 #1241: The grid input overview page doesn't show any data when importing power

DanielMcInnes, 2024-06-17:
Seen at "GRN - RS 3phase without EM"
On the overview page, the grid widget shows power being imported. When I click on the grid widget, PageAcIn is displayed, with all of the data fields empty.

[image]
[image]
[image]

blammit, 2024-06-18:
I thought there was another issue like this, but I can't find it...

blammit, 2024-06-18:
This is because the AC input on this system comes from a Multi RS system, not a grid or genset service, so showing the grid settings page in this case doesn't make sense. We could show PageMultiRs instead.

PR #1252 (2024-06-19): Use MultiRS settings as AC-in overview drilldown if needed
If the current AC input is from a Multi RS, show PageMultiRS when the AcInputWidget is clicked.

Fixes #1241
commit ead89449 (2024-06-19): Use MultiRS settings as AC-in overview drilldown if needed

If the current AC input is from a Multi RS, show PageMultiRS when
the AcInputWidget is clicked.

Fixes #1241
closed 2024-06-19 by blammit