victronenergy/gui-v2 #1249: Parent device page still shows the old name after renaming the device

jpetrell, 2024-06-18:
Open device in Device list settings. Go to Device. Rename the device name. Go back one step. Then go back to device list page.

Expected result: The new name is shown.

Actual result: While device list settings listing shows the updated name, the parent device page after back-stepping initially still displays the old name.

blammit, 2024-06-18:
Is this timing dependent? I couldn't reproduce it via wasm when renaming a solar charger or a battery.

jpetrell, 2024-06-19:
> Is this timing dependent? I couldn't reproduce it via wasm when renaming a solar charger or a battery.

I can reproduce on battery. Here is a video demonstrating the issue using digital input:
https://github.com/victronenergy/gui-v2/assets/2203667/e7c2b932-b6c2-4fe4-9749-0642fdd998df

The respective list item on the Device list page updates, but the device setting page title doesn't update unless you reopen the page.

blammit, 2024-06-20:
Oh sorry, I misunderstood. I thought you were referring to the item in the device page, rather than the page title.

blammit, 2024-09-25:
Passing the name as a binding (instead of as a static value) from the Device List should fix this for the most part.

PR #1533 (2024-10-02): Show updated device name when it is changed by user
From the Device List, if you open a device page and then open the "Device" sub-page and change its custom name, then step back to the device page, the page should show the updated device name.

Fixes #1249
commit b1201cd0 (2024-10-02): Show updated device name when it is changed by user

From the Device List, if you open a device page and then open the
"Device" sub-page and change its custom name, then step back to the
device page, the page should show the updated device name.

Fixes #1249
closed 2024-10-02 by blammit