victronenergy/gui-v2 #1254: Redirect user to HTTPS after selecting secured profile

jpetrell, 2024-06-19:
gui-v2 is adapted for the new security changes in #811. Changing the security profile in General settings reloads the web server. In most cases (secure->weak, weak->secure, secure->unsecure, weak->unsecure) when changing the security profile the browser automatically refreshes and reloads the gui-v2 successfully, but if you are using unsecure HTTP and enable the secured profile a redirect needs to be added.

Tried to implement redirect earlier, but just jumping to the new address failed because the web server hasn't finished reloading:
https://github.com/victronenergy/gui-v2/tree/jpetrell/https-redirect

chriadam, 2024-06-27:
This is needed for the security work.

jhofstee, 2024-07-05:
https://github.com/victronenergy/gui-v2/commits/jeroen/security-profile/ should fix this. It should reload the page when the security profile changed and check that the webserver is available again.

chriadam, 2024-07-11:
I will review.

mpvader, 2024-07-11:
make sure when wanting to test this, use the latest security build.

Right now, the latest one is `security8`; notes and announcement are here in Slack: https://victrondevelopment.slack.com/archives/C076X0MNMBR/p1720509538390829

once you get to work on this, @chriadam , make sure to read down from that message to see if there is a newer one.

chriadam, 2024-07-12:
Added some review comments, mostly LGTM but there are some small nitpicks.

jhofstee, 2024-07-22:
I update the branch to include the comments.

chriadam, 2024-08-02:
This has been merged and works according to my testing.  Closing as fixed.

closed 2024-08-02 by chriadam