victronenergy/gui-v2 #1256: Rethink how to handle gui-v2 translations with POEditor

jpetrell, 2024-06-19:
gui-v2 translations are based on ids, but the translation tool POEditor doesn't support those. Either drop the use of translation ids (i.e. qsTrId() -> qsTr()) or migrate to a different translation tool/service. Currently there are 100+ translations unhandled by POEditor, because of these. The issue is not critical, because missing translations get autofilled during build, but highlights architectural issue that should be at some point sorted out.

Current code probably has hundred places where the issue is worked around by adding (often low-value) translation comment or by moving the translation to CommonWords. This leads to recurring meta work and/or translation issues that sneak in the releases.

blammit, 2024-06-19:
Yes it's strange that we have CommonWords, since it leads to the use of the same text in different contexts, so in those cases we might as well be using qsTr() without translation hints. There are some attempts there to add context; for example, `open_status` has a value of "Open" with a hint of `//: Status is 'open'` to indicate it is not used as a verb, but with the amount of terms we have in CommonWords now, I am sure there are now many terms used in non-similar contexts.

blammit, 2024-11-25:
Needs further investigation into the issue mentioned by Joona to clarify what the issue actually is, as I don't recall myself.

Discussed with Chris that if we continue with qsTrId(), we should reduce the use of CommonWords.qml as it doesn't make sense to reuse translation ids where the context is different.

Also in deciding this, need to clarify with Dirk-Jan whether POEditor supports using different translation ids with the same Engineering English, i.e. it would be able to fetch a different translation in this case, if available. 

For example if you have:

```
//: Go back
//% "Back"
qsTrId("back_nav")

//: Side of object
//% "Back"
qsTrId("back_direction")
```

For the above, we would expect POEditor to fetch different translations if available. The translation catalog should have two different entries.

Example two:

```
//% "Back"
qsTrId("back_nav")

//% "Back"
qsTrId("back_nav_2")
```

Here, the translation catalog should have two different entries, but the translated strings should be identical, as no translation hints are provided.

blammit, 2025-01-06:
Something else to consider is, how third-parties would use translations. If third-party code needs to show translated text then does that mean they would need to know the translation id used for qsTrId()? Or, if there is new translatable text required in a third-party QML page, they would not be able to generate a new translation as the qm files are within the binary, and then the QML page may only show the engineering English. 

So this may be another advantage of using qsTr() instead: the third-party would not need to know the translation id, and also if the translation is not available, at qsTr() would return the untranslated engineering English, rather than showing a translation id in the UI.

MikeTrahearn-Qinetic, 2025-02-27:
There is a way to load .qm files on the fly for side-loaded QML (assuming they were generated by the third party). Perhaps this might be of interest. For the qsTr() case considering the "Back" situation, text-based translations have disambiguation which should be used in place of the hints.

MikeTrahearn-Qinetic, 2025-03-10:
Need to ascertain if third party untranslated qsTr() would pick up existing text-based translations because the context would come into play - that of the QML file - which may prevent this from happening.

MikeTrahearn-Qinetic, 2025-03-10:
# Researched answers to the above questions

## For the ID-Based case

**Two translations having the same id, and same //% engineering text**
(context is meaningless with id-based translations)

```
// someQMLFile.qml

    ...

    // a string which is re-used within this context
    // produces a single <message> elements with multiple <location> elements if enabled
    Text {
        //: Commonised Translation Instance 1
        //% "this string appears in multiple places in this context"
        text: qsTrId("id-duplicate-in-same-context")
    }

    Text {
        //: Commonised Translation Instance 2
        //% "this string appears in multiple places in this context"
        text: qsTrId("id-duplicate-in-same-context")
        // since the only thing that is different here is the comment,
        // these two ids end up being merged into a single <message> with multiple <locations>
        // and the comments are concatenated inside the <extracomment> with new lines and a dashed bar.
    }
```
**Two translations having the same //% engineering text, same //: comments but different ids**

These will create separate translation entries and will not be merged, even though their source text is identical.

In the case of custom QML being loaded at run-time, for ID-based translations (as has already been said) unless an existing ID is used, you will only see the engineering text - and then only if you lupdate/lrelease, otherwise you see the id!

## For the Text-Based case

**Two translations having the same source text in the same context**
(The context is the name of the QML file).

```
// someQMLFile.qml

    ...

    // a string which is re-used within this context
    // produces a single <message> element in the <context><name>someQMLFile</name>...
    // with multiple <location> elements if enabled
    Text {
        text: qsTr("this string appears in two places in this context")
    }

    Text {
        text: qsTr("this string appears in two places in this context")
    }
```

**Two translations having the same source text but in different contexts**

```
// QMLFile1.qml

    ...

    Text {
        text: qsTr("this string appears in different contexts")
    }

// CommonWords.qml

    Text {
        text: qsTr("this string appears in different contexts")
    }

```
The first ends up in `<context><name>QMLFile1</name><message> ...`
The second ends up in a **different** `<context><name>CommonWords</name><message> ...`

In the case of custom QML being loaded at run-time, for text-based translations, the qsTr() is not going be able to resolve an existing translation because there does not exist a `<context><name>...</name>` for it; it shall return the engineering text.