victronenergy/gui-v2 #1258: Levels page fails to display correct tank names

jpetrell, 2024-06-19:
Customer is [reporting](https://community.victronenergy.com/questions/292058/venus-os-v34026-available-for-testing.html?childToView=297192#answer-297192) that the Levels page fails to display tank names: "One small thing… running -29, I am using a Garnet Seelevel 709 RV-C for my tanks levels. In the level screen, the name are all the same for all tank, but the pictogram are good. Note that the propane tank is connected directly to the Cerbo. Also, I have no black tank but it is still showing."

[image]

Brief page and Devices list display correct names.

[image]
[image]

blammit, 2024-06-20:
Can repro with Boat/Motorhome 1 demo.

blammit, 2024-06-20:
Actually, what's happening here is that gui-v2 is showing the device "Product Name" as expected:

- Brief page: shows the tank type (e.g. "Fresh water")
- Device List: shows the tank type and device instance (e.g. "Fresh water (62)")
- Tank gauges (where this bug is reported): shows the tank `/CustomName`, or if that doesn't exist, the `/ProductName`

Here, because the first three tanks are from tank sensors, they don't have individual names: they do not have `/CustomName` values, and the `/ProductName` for all of them is "Tank sensor".

blammit, 2024-06-20:
> Also, I have no black tank but it is still showing.

Not sure what we can do about this one -- the tank must be still appearing under `com.victronenergy.tank.*` as a valid tank, if it is able to appear as a Black Water tank.

commit 4e4ee27c (2024-06-26): Use more meaningful names for Victron tank sensors

Use the tank type name instead of "Tank sensor" as the name.

Fixes #1258
PR #1276 (2024-06-27): Use more meaningful names for Victron tank sensors
Use the tank type name instead of "Tank sensor" as the name.

Fixes #1258
commit 7678e26a (2024-06-27): Use more meaningful names for Victron tank sensors

Use the tank type name instead of "Tank sensor" as the name.

Fixes #1258
commit 2e41dd73 (2024-06-28): Use more meaningful names for Victron tank sensors

Use the tank type name instead of "Tank sensor" as the name.

Fixes #1258
closed 2024-06-27 by blammit