victronenergy/gui-v2 #1261: The grid widget displays the wrong value for total power

DanielMcInnes, 2024-06-20:
This can usually be reproduced at 'GRN - RS 3phase without EM'.

[image]

In gui-v1, the total power displayed on the grid widget is always the sum of the 3 phases. In gui-v2, we seem to display 3x whatever the greatest single phase is. This becomes particularly obvious when a 3 phase is only importing power on a single phase.

[image]

blammit, 2024-06-20:
I think this bug is specific to AC inputs on Multi RS systems.

blammit, 2024-06-20:
Are the missing L2 and L3 values due to https://github.com/victronenergy/gui-v2/issues/1144?

That is, we're seeing

[image]

instead of

[image]

DanielMcInnes, 2024-06-20:
No, it's an unrelated issue. I originally fixed it on that branch, but decided it should be in a separate issue.

blammit, 2024-06-20:
Okay, is there a task for that one?

DanielMcInnes, 2024-06-20:
Here you go https://github.com/victronenergy/gui-v2/pull/1263

commit a2babb57 (2024-06-28): Fix calculation of total AC input power

Specific to AC inputs on Multi RS systems.

In ActiveAcInput, onPhaseValueChanged, instead of summing the power of the 3
phases, we were summing the updated phase power 3 times.

Now, we iterate over the phases correctly. We use a 1Hz timer (this is how
gui-v1 does it) to trigger the power calculation so that the screen isn't
updated top frequently.

Fixes #1261.
commit 1c8e3542 (2024-06-28): Fix calculation of total AC input power

Specific to AC inputs on Multi RS systems.

In ActiveAcInput, onPhaseValueChanged, instead of summing the power of the 3
phases, we were summing the updated phase power 3 times.

Now, we iterate over the phases correctly. We use a 1Hz timer (this is how
gui-v1 does it) to trigger the power calculation so that the screen isn't
updated too frequently.

Fixes #1261.
commit 4a3e1970 (2024-06-28): Fix calculation of total AC input power

Specific to AC inputs on Multi RS systems.

In ActiveAcInput, onPhaseValueChanged, instead of summing the power of the 3
phases, we were summing the updated phase power 3 times.

Now, we iterate over the phases correctly. We use a 1Hz timer (this is how
gui-v1 does it) to trigger the power calculation so that the screen isn't
updated too frequently.

Fixes #1261.
closed 2024-06-28 by DanielMcInnes