victronenergy/gui-v2 #1266: External grid metering should only be able to be set if an external meter device exists

chriadam, 2024-06-20:
Settings -> ESS -> Grid Set-point.  If you set the set-point to negative, then you allow discharging from battery to grid (i.e. feed-in from battery charge).  If you set it to negative, then you allow charging the battery from the grid (i.e. draw for battery charge).

However, if you have set the "Grid metering" setting to "External grid meter" without having a valid external meter device added to the system, then the battery will never charge from the grid even if you set a positive grid setpoint.

That option in the menu should be disabled if no external grid meter device exists in the system.

chriadam, 2024-06-20:
Marking as medium prio, not necessary for 1.0.0

blammit, 2024-11-25:
Check the gui-v1 behaviour for this and talk to Rein.

ReinvdZee, 2025-05-19:
I checked this on GUI-v1 on a system without external grid meter. Both options are there. When set to "External grid meter", changing the grid setpoint does not have any effect.

I agree that we should disable the external grid meter option there is none.

chriadam, 2025-05-20:
Thanks!  Taking this one.

chriadam, 2025-05-20:
Just to double check, I just need to determine if a device under .grid exists, which has /IsGenericEnergyMeter=1, and if so, enable that option (and otherwise, disable it), right?

I've noted that according to the dbus wiki the /IsGenericEnergyMeter path is also present on the .evcharger devices, but I assume I don't need to consider those ones since they're not related to the grid/battery interaction?  Or am I mistaken?

chriadam, 2025-05-20:
Also, can we assume that the external grid meter is connected to one of the two system ac inputs (i.e. I just need to check whether one of .system /Ac/In/<1|2> is a grid source, and if so, whether it has the /IsGenericEnergyMeter=1 value), OR are there some setups where some "inactive" (from the system's point of view) input is actually metered (and somehow also externally inverted?  in order to charge the battery with DC?) and should be considered also?

Maybe @izak could help me understand this also?

chriadam, 2025-10-30:
I can't remember what was decided about this one - could some clarification be added to this task?

chriadam, 2025-11-06:
@philipptrenz @ReinvdZee I think this needs more discussion in the meeting.  For now, I propose pushing it back to v1.3.0 at least.  Doing that now, we can undo my change if I am wrong.

philipptrenz, 2025-11-06:
From the comments @izak made, I think the decision if this setting is to be hidden should better be made by systemcalc, not in the UI. 

systemcalc has a mechanism to make a value "sticky". I.e. once there was a grid meter detected, the value on the path that indicate that stays at `1` until a device reboot. This is used to cope with PV inverters that lose connection over night and also suit this situation well.