victronenergy/gui-v2 #1267: Smart Shunt device issues

chriadam, 2024-06-20:
Was originally noted in the MultiRS issue, but it's unrelated to that:
1) settings -> device list -> smart shunt -> device -> "VE.Direct" text gets wrapped unnecessarily.
2) after disconnecting smart shunt the device is gone from device list rather than being cached in memory. (Is this really a bug? I guess, so, since it means the "Remove disconnected devices" button is useless if the device is always removed automatically?)

chriadam, 2024-06-20:
The first issue is high priority and should be fixed for 1.0.0.

The second issue is medium priority and is not necessary for 1.0.0, I believe.

ReinvdZee, 2024-06-20:
@chriadam Bullet 2 is an issue happening for all devices. Its [this](https://github.com/victronenergy/gui-v2/issues/1035) one.

DanielMcInnes, 2024-06-25:
I can't reproduce this issue now - in the smart shunt device page, 'VE.Direct' is displayed correctly. Running v3.40~30

DanielMcInnes, 2024-06-25:
@chriadam can I close this ticket?

chriadam, 2024-06-25:
Did you try connecting to Rein's device where we saw this issue, via WebAssembly / VRM?

DanielMcInnes, 2024-06-25:
yes, it's displaying correctly now.

chriadam, 2024-06-25:
This scares me a bit.  Did we change anything to do with this, recently?

ReinvdZee, 2024-06-25:
@chriadam @DanielMcInnes 

I can also not reproduce this issue right now. 
Yesterday, I've updated the system to V3.40~30, so it may be fixed now? 

I'll check some more devices with a VE.direct connected device today.

DanielMcInnes, 2024-06-25:
@ReinvdZee , do you mind if I downgrade your system, check the issue, then upgrade again?

ReinvdZee, 2024-06-25:
@DanielMcInnes 
>@ReinvdZee , do you mind if I downgrade your system, check the issue, then upgrade again?

Sure, go ahead. I'm not using it.

DanielMcInnes, 2024-06-25:
Thanks @ReinvdZee . I couldn't reproduce with v3.40~29 either. Ugh.

chriadam, 2024-06-27:
Removing high_prio label since we cannot reproduce the first issue. The second issue is low prio.

chriadam, 2024-06-27:
Actually, closing, as the second issue is covered by #1035

closed 2024-06-27 by chriadam