victronenergy/gui-v2 #1268: Generator: use new error scheme

ReinvdZee, 2024-06-21:
To support the new error scheme, update the gui to use `/Error/n/Id` instead of `/ErrorCode`. See related gui-v1 [commit](https://github.com/victronenergy/gui/commit/65e28c113e716603df488e177e1a4b45c0f2d8aa)

Note, the change on v1 is not merged yet because it will break things. So either store this change on a branch, or do it on request when we merge the rest of the generator changes.

blammit, 2024-06-27:
Blocked until available in Venus OS.

ReinvdZee, 2024-07-19:
Unblocked as the new error scheme implementation is merged per Venus OS v3.50~2

ReinvdZee, 2024-07-23:
We also need to implement the new [error scheme](https://github.com/victronenergy/venus/wiki/dbus#error-codes), which is a bit more than just renaming the path. 

With the new error scheme, the gui also needs to parse the error. E.g., a third-party device (genset in this case) reports something like `dse:e-4098` , which means error code 4098 from manufacturer DSE.
The mapping is in veutil, and the gui needs to use that. You probably need something similar to [this](https://github.com/victronenergy/gui/commit/9030a2d9ba686d1e5eb57a89a9b39ef7e5ba043d)

ReinvdZee, 2024-07-24:
I blocked this for now because we first need to work out the last specifics of the new error scheme.

mpvader, 2024-09-05:
How does this work now @blammit & @chriadam ?

Pls, to unlock this for releasing gui-v2: remove any fields that show genset errors, also for dc gensets, in the detailed pages of the genset. Users can see errors in the notifications; which is a path that already works now.

@ReinvdZee fyi: I want to not now start implementing parsing error codes into gui's again, the whole plan was to centralize that into Venus OS.

Later, if needed, we can make something for this; but for now, my above unlock idea is good enough wrt usability for customers.

cc @martinbosma & @jhofstee

mpvader, 2024-09-05:
ok decided on a slightly different plant: gui-v2 will look at all 8 paths, will show all the codes, and make it clickable and take user to notifications page where he can see the text explaining the code.

commit e00960d7 (2024-07-24): Update genset error code path

Use /Error/0/In instead of /ErrorCode
See 2089e6ae3c828d1ffd4f59d18b18b013f70cb596 in gui (v1)

Fixes #1268
PR #1346 (2024-09-11): Update genset error code path
Use /Error/0/In instead of /ErrorCode
See 2089e6ae3c828d1ffd4f59d18b18b013f70cb596 in gui (v1)

Fixes #1268
commit bfc99df1 (2024-09-11): Update genset error code path

Use /Error/0/In instead of /ErrorCode
See 2089e6ae3c828d1ffd4f59d18b18b013f70cb596 in gui (v1)

Fixes #1268
closed 2024-09-11 by DanielMcInnes