victronenergy/gui-v2 #1269: Parallel Lynx BMS: Show that Lynx BMS is waiting to connect to a parallel system

ReinvdZee, 2024-06-21:
See [this](https://github.com/victronenergy/venus/issues/1283) issue, we'd like to show that a Lynx BMS is waiting to connect to the system. Currently, the BMS summary just shows its Soc, voltage and current (0A, because its not connected). 

Instead, the following should be displayed: 

-  Summary (device list entry): "Waiting" + voltage delta + threshold 
-  BMS state: "Waiting" 

**BMS state**
When the BMS is in the waiting state, the path `/State` will be 18, which is a new state dedicated for this situation.

mpvader, 2024-06-21:
A delta? I thought we were going to show the absolute values.

ReinvdZee, 2024-06-24:
>A delta? I thought we were going to show the absolute values.

In the meeting of 04-06, we decided to show the voltage delta + threshold on Gui-v2. On V1 we show the system voltage.

ReinvdZee, 2024-07-02:
I've updated the original issue description and removed the blocked label. 

@blammit @chriadam @DanielMcInnes

DanielMcInnes, 2024-07-15:
Hi @ReinvdZee , I'm having trouble debugging my issue with the replay files you sent me. The 2 BMSes don't show up for me in gui-v2, the data for them is not published over mqtt. I think it will be easier to debug this issue on the 'Lynx smart BMS  parallel -  test setup' we were using last week, with a BMS in 'Pending' state. (On that cerbo, the 2 BMSes do show up for me via mqtt).

closed 2024-07-30 by DanielMcInnes