victronenergy/gui-v2 #1274: Overview page: grid widget: decimals aren't shown for power values

DanielMcInnes, 2024-06-25:
If each phase is 9999kW, the value displayed is 10MW, not 9.999MW
[image]
[image]

blammit, 2024-07-01:
Note:  for `W` values, this is the expected behavior, and there should be no decimals. However when the power is scaled to `kW`, `MW` etc. the decimal should be shown.

ReinvdZee, 2024-07-25:
We document the number of decimals of units [here](https://docs.google.com/document/d/1yzHSpZoeZ0WOJJPhjivbhhD_iz37PaZtnZ9GpqVPh2M/edit#heading=h.82agtgjmtjv)

chriadam, 2024-07-26:
Not sure this is a bug, as Joona specifically added a codepath to do this.  I recently expanded his code to make it more readable and maintainable, you can find that here: https://github.com/victronenergy/gui-v2/blob/main/src/units.cpp#L332

Basically, the precision is clipped to zero if we are using a scaled-up unit and the value is high enough.  It also does some anti-hysteresis stuff to ensure that it doesn't cycle between 10MW and 9.999 MW over and over, etc.

DanielMcInnes, 2024-08-20:
I don't think this is a bug either, closing the issue

closed 2024-08-20 by DanielMcInnes