victronenergy/gui-v2 #1275: SD card is not detected

blammit, 2024-06-26:
When an SDCard is installed and being used for VRM logging, not correctly detected. See screen shot from the v1 and v2 GUI.

[image]
[image]

ReinvdZee, 2024-07-15:
In gui-v1, the gui's backend connects to some dbus signals: https://github.com/victronenergy/gui/blob/master/platform/storage_events_dbus.cpp

Unmounting is done by requesting `vrmlogger` to do so: https://github.com/victronenergy/gui/blob/master/qml/MbMountState.qml

`VrmLogger` already also handles mount events; a storage medium will be used for logging when inserted; even without gui-v1 running. 

I tested this on an Ekrano GX with gui-v2 running locally and it seems to work properly (both local and wasm display correct values):

[image]

@blammit is this still an issue?

blammit, 2024-07-16:
@nmbath could you please re-test with the latest version?

blammit, 2024-08-13:
Thank you, will close this one then.

nmbath, 2024-08-23:
@blammit, I think this needs re-opening

Looking more closely at the screen shot I provided it is showing that Storage is internal and not external. I have posted on slack that there are issues.  https://victrondevelopment.slack.com/archives/C04RYU0UMDE/p1724420721369989

Summary:
GUI when accessing remotely does not display the same as when viewing on Touch 50.

Touch 50
[image]

Remote
[image]

Further and not sure this is a GUI thing!  If the SDCard is inserted it is found, following a reboot it is not detected.

blammit, 2024-08-26:
Re-opening for investigation.

blammit, 2024-08-30:
Discussed with @nmbath that he couldn't repro with the latest version.

closed 2024-08-30 by blammit