victronenergy/gui-v2 #1277: Temperature sensor names not displayed in the "Temperature control rules" menu.

ReinvdZee, 2024-06-26:
Reported on community: 

[image]

I reproduced this:
After configuring a relay as "Temperature", the submenu "Temperature control rules" shows up. This submenu shows the active temperature sensors, but all are just named "Temperature sensor". I changed the custom name of the temperature sensors, but it is not showing here. Like in the devicelist, this menu should show the `/CustomName` if set, and `/ProductName` if `/CustomName` is not set. 
[image]

blammit, 2024-06-30:
Currently `com.victronenergy.temprelay/Sensor/<sensor_id>/ServiceName` provides the service name for each temperature sensor under ``com.victronenergy.temperature.*`. (For example, if there's a sensor with the name `com.victronenergy.temperature.ruuvi_f00f00d00001`, the "sensor id" is `ruuvi_f00f00d00001`.)

On D-Bus, this `/ServiceName` is sufficient for locating the relevant `com.victronenergy.temperature.*` service for each sensor. However on MQTT, gui-v2 also needs a device instance to locate the service. This is a similar problem to https://github.com/victronenergy/gui-v2/issues/1025 where `/ActiveBmsInstance` was added to allow the device instance to be found for the BMS service.

@ReinvdZee Could you also add `com.victronenergy.temprelay/Sensor/<sensor_id>/ServiceInstance` or something like that, so that on MQTT gui-v2 can compose an id like `mqtt/temperature/<DeviceInstance>` to find the correct `ProductName` or `CustomName` for the service?

ReinvdZee, 2024-07-01:
@blammit 

I've [added](https://github.com/victronenergy/dbus-tempsensor-relay/commit/6e8d176b211b08fcfa8277b3bd433817cf3b1c2a) `com.victronenergy.temprelay/Sensor/<sensor_id>/ServiceInstance` to the service. 

Will be included in 3.50 beta.

blammit, 2024-07-05:
Thanks @ReinvdZee !

blammit, 2024-07-08:
Need to mark as blocked until the change to add `com.victronenergy.temprelay/Sensor/<sensor_id>/ServiceInstance` in VenusOS is released.

mpvader, 2024-07-24:
Hey @blammit , this is unblocked now, per Venus OS v3.50~5 its available.

blammit, 2024-07-25:
Thanks! Will look into adding this support.

PR #1358 (2024-07-30): Temperature control rules: show temperature sensor name on MQTT
In the temperature control rules, show the temperature sensor name as expected, when running on a MQTT backend.

On MQTT backends, the mqtt/temprelay/Sensor<sensor_id>/ServiceName value is not enough to locate the correct temperature service; it needs the service instance from
mqtt/temprelay/Sensor<sensor_id>/ServiceInstance as well, to assemble the full mqtt/temperature/<DeviceInstance> uid.

Fixes #1277
commit b72ab951 (2024-07-30): Temperature control rules: show temperature sensor name on MQTT

In the temperature control rules, show the temperature sensor name as
expected, when running on a MQTT backend.

On MQTT backends, the mqtt/temprelay/Sensor<sensor_id>/ServiceName
value is not enough to locate the correct temperature service; it
needs the service instance from
mqtt/temprelay/Sensor<sensor_id>/ServiceInstance as well, to assemble
the full mqtt/temperature/<DeviceInstance> uid.

Fixes #1277
closed 2024-07-30 by blammit