victronenergy/gui-v2 #1290: Display off timer setting does not work

blammit, 2024-07-01:
Should be able to set the display off timer from Settings > Display.

blammit, 2024-07-01:
Looks like `screenblanker.cpp` should respond to changes to `/Settings/Gui/DisplayOff` from the UI, just as `screensaver.cpp` does in gui-v1.

blammit, 2024-07-02:
Note: when implementing this, also hide the "Display off timer" setting on wasm builds. We already do this for adaptive brightness and brightness settings.

blammit, 2024-07-05:
Can't reproduce this issue, waiting for further information.

blammit, 2024-07-08:
Reported issue was that the timer options wouldn't work, and the displayed stayed on regardless.

DanielMcInnes, 2024-07-24:
I just changed the timer to 10 seconds on a wasm build, and after 10 seconds, my cerbo screen turned off. Cannot reproduce. I have made changes to hide the 'Display off timer' in wasm builds.

closed 2024-07-26 by DanielMcInnes