victronenergy/gui-v2 #1296: MultiRS devices not showing up in Device List

chriadam, 2024-07-04:
https://vrm.victronenergy.com/installation/394634/dashboard
device list is empty on main (only device in that system is multirs device, but it's invisible / not there?)
It shows up in gui-v1 but not gui-v2?

https://vrm.victronenergy.com/installation/394636/dashboard
device list contains three devices (external meter, BMS, etc) but cannot see the multirs device...
This one seems somehow broken in gui-v1 also?

chriadam, 2024-07-04:
Once this is resolved, we need to re-verify #1093 and #1077

ReinvdZee, 2024-07-04:
I've updated both systems to v3.40~34. 

On [RS 3phase with EM](https://vrm.victronenergy.com/installation/394636/overview), the AC system is back again, on gui-v1 and v2. I did not investigate, but I think some logic in systemcalc may have changed which handles `com.victronenergy.acsystem..`

On [RS 3phase without EM](https://vrm.victronenergy.com/installation/394634/overview), the AC system is not there. `com.victronenergy.acsystem..` is also not showing up in dbus, so that may be the issue. However, the multi's do show up (`com.victronenergy.multi.socketcan...`). I'm investigating what the issue is on this one.

ReinvdZee, 2024-07-04:
Actually, on the system without EM, `dbus-acsystem` is throwing an exception. I've opened a separate issue for this: https://github.com/victronenergy/venus/issues/1303

@chriadam @DanielMcInnes you can continue the verification of https://github.com/victronenergy/gui-v2/issues/1093 and https://github.com/victronenergy/gui-v2/issues/1077 on on [RS 3phase with EM](https://vrm.victronenergy.com/installation/394636/overview), please don't touch the other system for now.

DanielMcInnes, 2024-07-07:
While testing, the RS device would show up with the following code changes:
```

bool BaseDevice::isValid() const
{
	return !m_serviceUid.isEmpty() /*&& (!m_productName.isEmpty())*/ && m_deviceInstance >= 0;
}

```

DanielMcInnes, 2024-07-07:
While testing, we found that the RS device would show up if the check for `m_productName` was commented out:

```
bool BaseDevice::isValid() const
{
	return !m_serviceUid.isEmpty() /*&& (!m_productName.isEmpty())*/ && m_deviceInstance >= 0;
}

```

IIRC, @chriadam said the check should be:
```
bool BaseDevice::isValid() const
{
	return !m_serviceUid.isEmpty() && ((!m_productName.isEmpty()) || !m_customName.isEmpty()) && m_deviceInstance >= 0;
}

```
... but I don't think `m_customName` is implemented yet.

blammit, 2024-07-08:
Thanks, this is relevant to the discussion we had in the meeting where we weren't sure whether the current BaseDevice::isValid() test would make sense, as the check gui-v1 is slightly different. So I guess this confirms that we can't use ProductName as a firm test of whether the device is valid :)

commit dea6278e (2024-07-09): Consider a device valid if it has /CustomName instead of /ProductName

Some valid devices do not set /ProductName, but they do have a valid
/CustomName, so consider a device valid if either of these are set.

To check this in mock data, set /CustomName instead of /ProductName
where the name is not a proper product name.

Fixes #1296, #1233
PR #1310 (2024-07-09): Consider a device valid if it has /CustomName instead of /ProductName
Some valid devices do not set /ProductName, but they do have a valid /CustomName, so consider a device valid if either of these are set.

To check this in mock data, set /CustomName instead of /ProductName where the name is not a proper product name.

Fixes #1296, #1233
commit 3744aaf4 (2024-07-09): Consider a device valid if it has /CustomName instead of /ProductName

Some valid devices do not set /ProductName, but they do have a valid
/CustomName, so consider a device valid if either of these are set.

To check this in mock data, set /CustomName instead of /ProductName
where the name is not a proper product name.

Fixes #1296, #1233
closed 2024-07-09 by blammit