victronenergy/gui-v2 #1298: Reconnection loop & black screen with Fixefox on Android

jepefe, 2024-07-04:
This can be easily reproducible on Android:

1. Open the gui-v2 with Firefox
2. Send Firefox to the background and do something else or switch off the screen for a while.
3. Open Firefox
4. The rings animation is displayed but enters into a reconnection loop and sometimes just shows black screen

I suspect that the app is paused and then when resumed the websocket connection is broken.  If this can't be solved somehow, would be nice to detect that the app is in this state and reload the webpage.

The Android dedicated display uses Firefox GeckoView so would be nice to have this fixed for v1.

mpvader, 2024-07-11:
Chris will do some testing tomorrow.

If it becomes all complicated, ie emscriptem/Qt upgrading etc etc doesn't fix it, then we won't fix this for v1.0.0

chriadam, 2024-07-12:
Can you provide information about which version of firefox you see this, and precise repro instructions?

I cannot reproduce with Firefox on my Android phone.
Version information:
128.0 (Build #2016030615)
hg-14b32d530926+
GV: 128.0-20240704121409
AS: 128.0.1
2024-07-04T18:19:30.039769603

I tested connection to local cerbogx nginx via wifi, and also to VRM.
For local cerbogx connection, it always reloaded the page properly.
For VRM connection, after reactivating the tab, the gui-v2 overlay thing was gone, I had to click click the menu hamburger then click gui-v2 to reactivate it.  That worked.
With a shorter interval of inactivity for that tab, I was able to switch back to it and see that the gui-v2 overlay remained (but had transitioned to transport error state), but in this case after a few seconds it reconnected properly.
Even activating a different tab in firefox, then switching from wifi to mobile data, then reactivating the vrm tab with gui-v2 open, worked. (It showed a transport error, then after a few seconds reconnected and reloaded the UI properly.)

DanielMcInnes, 2024-07-12:
I can't reproduce this either (firefox & android). I tried wasm builds in a variety of scenarios (local cerbo via wifi, remote cerbo via URL, and launching gui-v2 from the vrm website).

mpvader, 2024-07-15:
Also @andres84ve tried to reproduce this; and can't.

This is on hold until @jepefe can reproduce it.

chriadam, 2024-08-02:
This seems to be quite device specific.

Jesús was able to reproduce with the following:

Android 11: Motorola Moto G5 Plus & Samsung Galaxy M30s
Android 13: Galaxy Tab A9 & Lenovo Tab M10 Plus 3rd Gen
Android 14: Samsung Galaxy S23

To reproduce:
- Open Firefox
- Go to http://your_venus_ip/gui-beta
- Play with the gui: Just to make sure it is working fine( Probably not required)
- Send the Firefox app to the background by opening other app.
- Wait: The amount of time is a bit random,  just leave Firefox in the background and use your phone as usually. 30 minutes should be enough, but I've seen the issue when waking up the phone after a couple of minutes only.
- Bring Firefox to the foreground, then the rings animation is executed and two things can happen after it: Reconnection loop, or Black screen.

However, I was unable to reproduce with my Oppo A52 running Android 11.

Daniel was eventually able to reproduce on his device, after setting the device's battery setting to power saving mode and setting screen-turn-off-timeout to 15 seconds.

We will investigate further.

DanielMcInnes, 2024-08-05:
[Uploading wasm-bug.txt…]()

DanielMcInnes, 2024-08-06:
new log file:
[console-export-2024-8-6_16-31-26.txt](https://github.com/user-attachments/files/16506241/console-export-2024-8-6_16-31-26.txt)

DanielMcInnes, 2024-08-06:
new log file:
[console-export-2024-8-6_17-57-55.txt](https://github.com/user-attachments/files/16507202/console-export-2024-8-6_17-57-55.txt)

DanielMcInnes, 2024-08-06:
success:
[console-export-2024-8-6_18-27-38.txt](https://github.com/user-attachments/files/16507600/console-export-2024-8-6_18-27-38.txt)

chriadam, 2024-08-06:
@jepefe are you able to test that my fix https://github.com/victronenergy/gui-v2/pull/1392 fixes the issue for you?
I've attached a patched-and-packed wasm build of that branch for you to test.
[gui-beta.tar.gz](https://github.com/user-attachments/files/16507803/gui-beta.tar.gz)

jepefe, 2024-08-06:
@chriadam I'll do it later today and let you know.

jepefe, 2024-08-06:
@chriadam The fix seems to work, or at least I haven't been able to reproduce the issue.

chriadam, 2024-08-08:
Thanks!  PR has been merged to main.  Closing as fixed.

closed 2024-08-08 by chriadam