victronenergy/gui-v2 #1301: Grid drilldown widget shows incorrect firmware version

chriadam, 2024-07-05:
Using ESS demo, clicking the Grid widget from Overview page leads to a drilldown page which shows "Firmware version: v0.02" but gui-v1 shows "Firmware version: 2" and dbus-spy also shows value: "2".

chriadam, 2024-11-01:
I looked into this, and it seems that what is happening is as follows:

1) the value of the setting `com.victronenergy.grid.cgwacs_ttyUSB0_di30_mb1/FirmwareVersion` is `2` (a number).
2) in gui-v1 this is just displayed directly (see qml/PageDeviceInfo.qml)
3) in gui-v2 we pass this value to the versionText() function within FirmwareVersion.qml, which has some rules for converting version numbers into strings (which are then displayed in the UI).  It documents some of the patterns which it uses during conversion, and the case we are hitting for this bug is:

a) the value is a number (2), and not a string ("2")
b) so it assumes that the version number is a hex code, so converts to hex
c) this hex value is of the form 0x00000A
d) so it returns the string: "v0.0A"

--

I guess the question is: is the logic described above only intended for some specific type of version, and not in this (grid) case?

In gui-v1 there is e.g. `return firmwareVersion.value >= 0x109` in `qml/PageSolarCharger.qml` which suggests that at least some devices assume that the version is given as a number which is meaningful when converted to hex.

Are there some specific rules for when the number/hex/string conversion rules should be applied?  @jepefe perhaps knows?

jepefe, 2024-11-11:
Gui-v1 uses the GetText which provides the correct version format while gui-v2 GetText is not available so the raw value from DBus is used. 

@izak Is it possible to change dbus-cgwacs to report `/FirmwareVersion` with the expected value? Or will it break something else?

izak, 2024-11-11:
I assigned this to myself. For CG meters the firmware version, which is really just a 16-bit integer, is published as an integer on dbus, but really for third party meters where we don't control the numeric->text translation, it is better that the meter publishes its firmware version as text.

jhofstee, 2024-11-18:
The GetText version is included in Venus candidate, version v3.52~6.

blammit, 2024-11-21:
Fixed by https://github.com/victronenergy/gui-v2/pull/1743

closed 2024-11-21 by blammit