victronenergy/gui-v2 #1302: Solar Yield with PV Inverter set as Output displays incorrectly on overview page

chriadam, 2024-07-05:
Using ESS demo:
- click the Solar Yield widget, click "Position" and change from AC Input 1 to AC Output, then backstep.
- what actually happens: the AC Loads widget value increases from 500W to 1500W
- what should happen: the solar yield widget should shift over to the "Outputs" side of the Overview page, but AC Loads value remains

Technically, the solar yield is PROVIDING power, i.e. it should count as a NEGATIVE AC Load
Not sure how to display this properly...  Need input from @Serj and @jepefe

chriadam, 2024-07-05:
Similar issue seen with Boat/Motorhome Demo 1 - this one has an Alternator suppling 1600W, a DC Load drawing 600W, and AC Loads drawing 335W, and yet the Battery is showing draw of ~1300W.

To me, it looks like the battery value being displayed is calculated something like ((1600 - 600) + (335)) which ... doesn't make sense.  The value should really be: ((1600 - 600) - (335)) i.e. the battery is charging and drawing ~700W to charge the battery.

I am saying that this is "similar" to the original issue, because I believe that the Alternator is being treated as an "output position" sort of thing.  I might be wrong.

DanielMcInnes, 2024-07-07:
marking this one as blocked pending feedback from serj and @jepefe

ReinvdZee, 2024-07-11:
Behavior with the ESS demo (running on Venus OS 3.40~37). 

**Note, the numbers may not always add up because part of the numbers are calculated and others are measured. Since this is a dbus recording, the measured ones may be off when the system setup changes. The default position for the PV inverter is "AC input 1"**

1. Click on solar yield on the overview page
   - There are two solar devices, a Solar charger and a PV inverter. The PV inverter delivers ~1000W.
3. Click on the PV inverter
4. Set "Position" to "AC input 1" (already the case with the ESS demo)
[image]

5. Go back to overview page. 

Now the solar yield is on the left of the overview. 
[image]

_To me, it looks like the number don't add up, but that may be due to the simulation. Can be investigated separately_

VRM also shows the PV inverter as source on the left:
[image]

6. Now, to showcase the issue, set "Position" to "AC output"
Now, the solar yield hasn't changed. The only thing that changed is the value the AC loads widget displays. It went from ~500W to ~1500W (so it added the PV inverter output of ~1000W to the load of ~500W) So the PV inverter is connected to the inverter/charger **input** on the overview, but its actually connecting on the inverter/charger **output**, supplying the AC load directly. 

[image]

On VRM, the PV inverter is actually moved over the right: 
[image]

Also on Gui-v1, the PV inverter is moved over to the right side and connected to the inverter/charger output / critical loads.
[image]

Aside from the numbers being off, is there a better way to show in the overview that the PV inverter is **not** supplying grid but is supplying the AC output directly? I don't think it is clearly displayed right now. 

@jepefe.

jepefe, 2024-07-11:
I've checked this, and the discrepancy in the numbers is indeed due to the simulation mode. The VE.Bus Inverter data is a replay of static values and does not react, so SystemCalc assumes that the PV inverter production is entirely used by the loads.

Unlike GUI v1, where the overview attempts to be a 1:1 schematic representation of the system, in GUI v2, the current flows from left to right, with sources on the left and consumers on the right.

Since the PV is always a power source, it must be kept on the left, regardless of where it is connected. We are not going to change this. 

I think the "problem" here is that GUI v1 does a better job for this specific case, but overall, the current flow approach in GUI v2 is better and more flexible.

ReinvdZee, 2024-07-11:
@jepefe So we leave it as is?

jepefe, 2024-07-11:
Unless I'm overlooking something, yes, there is not real issue.

blammit, 2024-07-19:
@chriadam @ReinvdZee should this one be closed then?

chriadam, 2024-07-19:
Yes, thanks.  Closing as resolved.

closed 2024-07-19 by chriadam