victronenergy/gui-v2 #1303: Solar Yield drilldown page from overview should display Error Number as well as Error String

chriadam, 2024-07-05:
ESS Demo, Overview Page, Solar Yield drilldown, Error list entry should display number (#0) as well as string ("None").

chriadam, 2024-07-05:
When changed from 0 to 1 via dbus-spy, the number displays as red, but there is no description.

chriadam, 2024-07-05:
Same issue applies to the MPPT (PV Charger) drilldown.  Error code isn't shown, or description isn't showing.

blammit, 2024-07-12:
@chriadam This appears to be working fine for PV chargers, or at least on [this](https://vrm.victronenergy.com/installation/183123/dashboard) if I go to Overview > Solar drilldown > Some PV charger > Alarms and Errors, I see this:

[image]

blammit, 2024-07-12:
@jepefe @ReinvdZee can [charger_error from veutil](https://github.com/victronenergy/veutil/blob/master/inc/veutil/qt/charger_error.hpp) be used to translate PV Inverter error codes into text descriptions? In [charger_error.cpp](https://github.com/victronenergy/veutil/blob/master/src/qt/charger_error.cpp) I can see some inverter-specific errors like `CHARGER_ERROR_INVERTER_OVERLOAD = "Inverter overload"`, but just want to double check.

commit 60d4fdf6 (2024-07-15): Show detailed error string on PV inverter page

Use the same error text that is shown when clicking on a PV Inverter
from the Device List.

Fixes #1303
closed 2024-07-15 by blammit