victronenergy/gui-v2 #1307: Alternator getting assigned current consumption

msigman, 2024-07-07:
When an external regulator such as Wakespeed WS500 is connected to the Cerbo with Lynx BMS, and the Wakespeed is turned ON but the engine is OFF (so not producing current), the Cerbo is assigning "negative" current to the alternator.  See attached photo for example.  In this photo we had 331W of DC consumption, of which 321W was assigned to the alternator when turned on.

Recommended fix: Do not allow alternator current to drop below 0 (go negative) when calculating DC loads.
[image]

blammit, 2024-07-16:
@jepefe @ReinvdZee FYI

chriadam, 2025-11-06:
Pushing back to v1.3.0.  Raising prio to medium as this needs to be discussed early in the v1.3.0 cycle.

ReinvdZee, 2026-09-10:
The backend service should be responsible for clamping the current value to 0A.

ReinvdZee, 2026-09-15:
This will be solved on the backend, see: https://github.com/victronenergy/venus/issues/1681.

Let's keep this issue open until the backend changes are in and verified

nmbath, 2026-09-15:
An alternator can consume current and should be accounted for,. Depending on the situation the alternator controller is being powered and feeding a field drive, even when the alternator is not spinning, or spinning and not generating. Fixing this in the driver is wrong as the current is negative.

nmbath, 2026-09-15:
I would also add that the system is displaying what it has received from the Alternator controller, in this case a negative current, so it is true.

In this case the wakespeed is reading a negative current from the shunt it is connected too; The wakespeed is then reporting this over CAN, the vecan-dbus driver is then reading this and pushing it to dbus. So why is it wrong?

At ideal my 24v Alternator controlled by a Wakespeed 500 Pro draws circa 110W. In a catamaran with two controllers, that would display as an aggregate of circa 220W.

Other setups could draw more, I do not have a Wakespeed 500 (non pro) to know what it draws at idle. Also different alternators themselves may have different field setup’s which may cause a higher current,

ReinvdZee, 2026-09-15:
Thanks @nmbath,

I realize you're right. After dicussion with Izak and Martin, we decided to not change anything and close the issue. All we should do is display the values as reported by the alternator. The alternator controller is responsible for reporting the current and power value correctly and if that is negative, then it seems to be consuming power.

closed 2026-09-15 by ReinvdZee