victronenergy/gui-v2 #1319: Display individual Lynx Smart BMSs in the device list

DanielMcInnes, 2024-07-12:
Gui-v2 only displays the totals for Lynx Smart BMS. Gui-v1 displays the individual BMSes as well.

[image]
[image]

blammit, 2024-07-12:
Screenshots are for [this](https://vrm.victronenergy.com/installation/323213/overview ) installation.

blammit, 2024-07-12:
On [the installation](https://vrm.victronenergy.com/installation/323213/overview), gui-v2 has four battery services under `com.victronenergy.battery.*`, as follows:

[image]

The second one ("lynxparallelundefined") is probably a bug that can be ignored. The problem is that only the `com.victronenergy.battery.lynxparallel2` service appears in MQTT under `mqtt/battery/*`; the `com.victronenergy.battery.socketcan_can0_*` services do not, so they are not accessible by gui-v2 on wasm.

@ReinvdZee  would you know why this is happening? Is there something special about those battery services? I found that [this installation](https://vrm.victronenergy.com/installation/402872/overview) has similar `com.victronenergy.battery.socketcan_can0_*` services, but they are accessible under `mqtt/battery/*` as expected so they do appear in gui-v2 on wasm, so I'm not sure what's happening here.

blammit, 2024-07-12:
@DanielMcInnes mentioned the installation is playing a replay, so perhaps that's the issue?

DanielMcInnes, 2024-08-20:
Closing issue, this has been fixed somewhere else, individual Lynx smart BMSes are now displayed individually

[image]

closed 2024-08-20 by DanielMcInnes