victronenergy/gui-v2 #1324: Add alternator temp to Overview view tile

msigman, 2024-07-12:
Add the alternator temp to the Overview tile, like the battery temp.

[image]

blammit, 2024-07-16:
@jepefe fyi. Are there any plans to expand the details shown in the overview boxes?

ReinvdZee, 2026-09-10:
@ReinvdZee will check with Serj if this is still wanted

ReinvdZee, 2026-09-15:
@ybott-qinetic 

Discussed this with Serj and we decided to show the alternator temperature on the top-right corner of the widget if available. Please make sure to show the value in the GX-wide temperature unit. According to the [dbus wiki](https://github.com/victronenergy/venus/wiki/dbus#alternator-and-dcsource), the temperature is reported under the path `/Dc/0/Temperature` in degrees centigrade. Do not show anything if the path is not there.