victronenergy/gui-v2 #1325: Add solar & wind priority feature

jepefe, 2024-07-13:
The Solar & Wind priority feature is available for VE.Bus inverter/chargers, prioritizing renewable power sources to charge the batteries over using the AC input.

The feature is enabled in the inverter, and the GUI only needs to display the system's status and provide a button to override the normal operation and force the batteries to charge.

When Solar & Wind priority is enabled, the current GUI shows an icon on the battery. If charging is forced, this is indicated by a red strip on the battery icon.

The button to force the charge is available on the VE.Bus inverter page.

Serj needs to provide a design proposal for this feature.

All the details for implementation can be found in this issue:
https://github.com/victronenergy/venus/issues/1052

Documentation:
https://www.victronenergy.com/live/ve.bus:solar-and-wind-priority
https://www.victronenergy.com/media/pg/Solar_&_Wind_Priority/en/index-en.html

mpvader, 2024-07-14:
Hey Jesus, this one is nearly impossible to have ready before we want to release v1.0.0.

@blammit there is not much you can do on this until there is a design.

blammit, 2024-11-25:
Changing to high as 1.0.0 has been released and we need this for feature parity.

blammit, 2025-01-08:
Design available in slack: https://victrondevelopment.slack.com/files/U021EAE390Q/F07NB9XG6RF/solar___wind_priority.pdf

DanielMcInnes, 2025-01-14:
figma: https://www.figma.com/design/GjCtz3A3jqSGvyjVvOm6RZ/GX-Touch-design?t=HI8qgDGlsjA3CAmn-0

commit 2ef83e68 (2025-01-17): Add solar & wind priority feature

Fixes #1325
commit 091a19d7 (2025-01-17): Add solar & wind priority feature

Fixes #1325
commit 7057aeae (2025-01-17): Add solar & wind priority feature

Fixes #1325
commit 06fed4b7 (2025-01-17): Add solar & wind priority feature

Fixes #1325
closed 2025-01-26 by DanielMcInnes