victronenergy/gui-v2 #1328: Show zero decimal places for Watts (e.g. in AC Input overview widget)

chriadam, 2024-07-15:
Don't show decimals for Watts.
Show decimals for KiloWatts.

Check what the rules is for switching between watts and kilowatts (I think it's based on the available space to draw text, but usually the boundary will be 9999 Watts -> 10 kW, I believe - but double check this).

chriadam, 2024-07-15:
re: the rules for switching between watts and kilowatts: it's hardcoded to be a 10k boundary.  Code follows:
```
			// Kilo scale has 10k limit, other scales grow by 1k
			qreal multiplier = wasPreviousScale ? 0.9 : 1;
			if (scale == VenusOS::Enums::Units_Scale_Kilo) {
				multiplier = wasPreviousScale ? 9 : 10;
			}

			return qAbs(value) >= multiplier*qPow(10, 3*scale);
```

I guess this was discussed with Serj and it was decided that it made the most sense.  This can be revisited if required.

chriadam, 2024-07-15:
For the first / main issue, of showing decimals for Watts - actually, it will never show a decimal for Watts unless the value is between -1 and 1.  A code comment says:

```
	// If value is between -1 and 1, but is not zero, show one decimal precision regardless of
	// precision parameter, to avoid showing just '0'.
```

But if you would prefer to just show 0, we can do this for Watts, of course.  I will create a PR which does this, for discussion.

chriadam, 2024-07-15:
https://github.com/victronenergy/gui-v2/pull/1330 has this PR for discussion.

DanielMcInnes, 2024-07-23:
@chriadam can this issue be closed now that #1330 is merged?

chriadam, 2024-07-23:
Yes, thanks, closing as fixed.