victronenergy/gui-v2 #1334: Set root password

mr-manuel, 2024-07-15:
When I try to set the root password, I get the message "not yet implemented".

mr-manuel, 2024-07-15:
Seems to work with the Venus OS `security8` release and GUIv2 `v0.3.5`, but not with Venus OS `v3.40~39` and GUIv2 `v0.3.6`.

## Venus OS `security8` release and GUIv2 `v0.3.5`
[image]

## Venus OS `v3.40~39` and GUIv2 `v0.3.6`
[image]

mpvader, 2024-07-15:
Hey @mr-manuel 

This is intended to only work with the security branches.

and thus not yet plain v3.40~39

mpvader, 2024-08-02:
Duplicate with #1384

closed 2024-08-02 by mpvader