victronenergy/gui-v2 #1335: Time zone list

mr-manuel, 2024-07-15:
It seems very strange to see an entry `Etc`. Better would be `Other` or `Others`.

Settings -> Date & Time -> Time zone

[image]

blammit, 2024-08-21:
This is a direct copy of the list from gui-v1 so this is unlikely to change at the moment.

blammit, 2025-04-02:
@jhofstee Is it okay to change "Etc" to "Other" here? Will that cause any issues with changing the timezone in the lower layers?

mr-manuel, 2025-04-03:
Renaming `Etc` does not matter regarding `venus-platform`, but we need to find a better name for it.

mpvader, 2025-04-03:
As discussed in the meeting today, Chris will take care of this.

The strings supported by Venus OS are these:

```
root@ekrano:~# ls /usr/share/zoneinfo/Etc/
GMT        GMT+1      GMT+11     GMT+2      GMT+4      GMT+6      GMT+8      GMT-0      GMT-10     GMT-12     GMT-14     GMT-3      GMT-5      GMT-7      GMT-9      Greenwich  UTC        Zulu
GMT+0      GMT+10     GMT+12     GMT+3      GMT+5      GMT+7      GMT+9      GMT-1      GMT-11     GMT-13     GMT-2      GMT-4      GMT-6      GMT-8      GMT0       UCT        Universal
```

chriadam, 2025-04-10:
Merged to main, closing as fixed.

closed 2025-04-10 by chriadam