victronenergy/gui-v2 #1340: When a battery is created with an invalid serviceUid, it is never destroyed

DanielMcInnes, 2024-07-16:
`mqtt/system/0/Batteries` is updated regularly, perhaps when voltage / current changes? If  the reported batteries are not already part of the model, new Battery objects are created, and are then only added to the model when they become valid, which never happens in some situations. A few seconds later, `mqtt/system/0/Batteries` is updated again, and more invalid Battery objects are created.

DanielMcInnes, 2024-07-16:
I have only seen this happen on a simulated system, that reported batteries via mqtt/system/0/Batteries that didn't exist. However, if it did happen on a real system, it would be tricky to track down. I think gui-v2 should be able to cope with something like this without leaking.

blammit, 2024-07-22:
Real devices are unlikely to have an invalid service uid, so setting this to low priority.

blammit, 2024-08-05:
@DanielMcInnes  With the changes for https://github.com/victronenergy/gui-v2/issues/1357, all Battery objects generated from `mqtt/system/0/Batteries` are now deleted when when the Battery drilldown page is popped. I think this is sufficient to ensure they are cleaned up; is this okay?

DanielMcInnes, 2024-08-05:
@blammit , not really. We could be on any page, and every time battery current changes (I saw this happen about every 2 seconds) we create new battery objects. If they never become 'valid', they leak

blammit, 2024-08-05:
They shouldn't be created anywhere except the Battery list page now. This is a change from before, when they were created as part of the `Global.batteries.model` generation.

DanielMcInnes, 2024-08-05:
oh right, that should be ok then

closed 2024-08-06 by blammit