victronenergy/gui-v2 #1342: RS control pane

mpvader, 2024-07-16:
when accessing the on/off/charger-only for the RS, I encountered below pop-up.

Should Close be renamed to Cancel?

[image]

DanielMcInnes, 2024-07-16:
I think 'Cancel' would be better than 'Close', as it's more like what Windows does, and people are used to it. I personally would prefer to change 'Set' to 'OK' for the same reasons, but I don't know how everybody else feels about that.

DanielMcInnes, 2024-07-16:
@mpvader , re. "Should Close be renamed to Cancel?", it is unclear to me whether you want 'Close' renamed to 'Cancel', or you just want feedback.

mpvader, 2024-07-17:
Hey @DanielMcInnes , I wasn’t sure; but yes please change Close to Cancel.

Set / OK is perhaps a bit less trivial - I don’t know. Anyway lets leave that as is.

DanielMcInnes, 2024-07-17:
@mpvader , do you just want it changed for this specific dialog, or for all dialogs that have 'Close' and 'Set' buttons (there are many)? I will make a list of affected dialogs if you want.

commit 0d56fcb6 (2024-08-07): Change the default dialog reject text from 'Close' to 'Cancel'

Fixes #1342
PR #1394 (2024-08-07): Change the default dialog reject text from 'Close' to 'Cancel'
Fixes #1342
commit d614a0b3 (2024-08-07): Change the default dialog reject text from 'Close' to 'Cancel'

Fixes #1342
closed 2024-08-07 by DanielMcInnes