victronenergy/gui-v2 #1350: Add QR code generation support to gui-v2

chriadam, 2024-07-24:
For a third-party vendor (Tailscale) integration into VenusOS, QR code support is needed.  The settings page which will be integrated into gui-v2 must be able to generate a QR code based on some system information.

So, possible solution:
1) import qzxing (Apache 2.0 license) as a git submodule and build it as a dependency within gui-v2
2) ensure that its types are registered into appropriate QML namespace for use within gui-v2 settings pages code
3) ensure that it is built statically for WebAssembly builds.

My understanding is that this is needed for v3.50 release i.e. 1.0.0 milestone.

Should discuss all of these details with Matthijs at the meeting on Thursday to double check that I understood the requirements correctly.

kwindrem, 2024-07-24:
I don't know if you have seen the work I did with Tailscale integration. My TailscaleGX package is a wrapper around the tailscale mechanism providing a user interface and a connection state machine that reduces the steps needed by a user to setup tailscale. It is lacking the QR code interface for providing the URL to establish the connection but it otherwise functional

mr-manuel, 2024-07-24:
@kwindrem TailScale will be integrated into Venus OS. You can check my fork of your repo and test it.

mr-manuel, 2024-07-24:
Here a link to the page which needs the QR code generation: https://github.com/victronenergy/gui-v2/blob/mr-manuel/tailscale/pages/settings/PageSettingsTailscale.qml

blammit, 2024-07-25:
This is a nice to have for 1.0.0.

mr-manuel, 2024-07-25:
@mpvader would it maybe be better, if a cpp class or the Venus OS plattform can generate a QR code, which can then be used as a base64 image string via MQTT? I read somewhere what also in the MFD app QR code generation would be useful?

mpvader, 2024-07-25:
@mr-manuel maybe; but I don't think its necessary to add interdependencies like that; next we need to change the size of the QR code depending on where its shown, and an API is built, etc etc; while at the same time its should be trivial to make such thing with QT. And same to make a QR code with html5; trivial as well I expect. but if not - @chriadam ofcourse welcome to flag that

mr-manuel, 2024-08-08:
Temporary I added the generation to the Tailscale script, else there is no QR code for GUIv1.

See https://github.com/mr-manuel/venus-os_TailscaleGX/commit/19e9d9c4cced12917ee1b51de31c26d84cc4cc92 and the needed dependencies https://github.com/mr-manuel/venus-os_TailscaleGX/commit/5c549d8a75b47924b6f08a059f4cdfe46c5002ae.

Maybe this could stay like this?

DanielMcInnes, 2024-09-04:
Adding 'blocked' label until we decide whether to add a victron git repo for qzxing, or apply a patch. This question will be resolved at the next meeting.

mr-manuel, 2024-09-05:
I add a short reminder here to enable the QR code part for the Tailscale page after this is solved.

https://github.com/victronenergy/gui-v2/blob/30250661025cd2cb37af5e6dc3bbfa2e403c3403/pages/settings/PageSettingsTailscale.qml#L8-L9

https://github.com/victronenergy/gui-v2/blob/30250661025cd2cb37af5e6dc3bbfa2e403c3403/pages/settings/PageSettingsTailscale.qml#L175-L197

mpvader, 2024-09-05:
@martinbosma can you make a repository for this? thanks.

martinbosma, 2024-09-06:
@DanielMcInnes https://github.com/victronenergy/qzxing

DanielMcInnes, 2024-09-09:
Hi @martinbosma , I don't seem to have the permissions to push to that repo
```

dmcinnes@dmcinnes-System-Product-Name:~/git/victron/gui-v2/src/qzxing$ git push --set-upstream origin 1350-add-qr-code-generation-support-to-gui-v2
remote: Permission to victronenergy/qzxing.git denied to DanielMcInnes.
fatal: unable to access 'https://github.com/victronenergy/qzxing/': The requested URL returned error: 403
dmcinnes@dmcinnes-System-Product-Name:~/git/victron/gui-v2/src/qzxing$ 

```

martinbosma, 2024-09-09:
@DanielMcInnes Should be fixed

DanielMcInnes, 2024-09-09:
> @DanielMcInnes Should be fixed

Thanks Martin, looks good

commit 45e909aa (2024-08-14): Add QR code generation support

Fixes #1350
commit 4926ee31 (2024-09-09): Add QR code generation support

Fixes #1350
commit c5cb8a8b (2024-09-09): Add QR code generation support

Fixes #1350
commit 248f5502 (2024-09-16): Add QR code generation support

Fixes #1350
closed 2024-09-16 by DanielMcInnes