victronenergy/gui-v2 #1357: Device List should show all batteries under com.victronenergy.battery.*

blammit, 2024-07-25:
https://github.com/victronenergy/gui-v2/issues/1224 changed the battery list to be generated by `com.victronenergy.system/Batteries` instead of `com.victronenergy.battery.*`. However, this affects the batteries shown in the Device List as well. 

We actually need two separate battery lists, like this:

- When clicking on the Overview battery, show a list of all batteries in com.victronenergy.system/Batteries (if there is more than one)
- In the Device List, show all battery services under com.victronenergy.battery.*

mpvader, 2024-07-26:
hey @blammit , this is one high prio rather than medium; since the data/menus of those batteries are now effectively no longer reachable unless worked around by setting things up in settings -> system -> battery measurements.

blammit, 2024-08-01:
Can be tested on [this](https://vrm.victronenergy.com/installation/133622/dashboard) installation. There are three com.victronenergy.battery.* services, but only one battery in the `com.victronenergy.system/Batteries` list. So, the Device List should show the three batteries, but only one should be accessible via the Overview Battery drilldown.

blammit, 2024-08-01:
@mpvader  I noticed on [this](https://vrm.victronenergy.com/installation/366384/dashboard ) installation that `com.victronenergy.system/Batteries` may include `genset` services. In this case, would you expect the battery drilldown list to show the details that are typically shown for genset services, like this:

[image]

Because the problem is, at the moment, the battery drilldown list tries to show the battery details like `/Soc`, `/Dc/0/Voltage` etc. but that is not present for genset services, so it looks like this in the drilldown list:

[image]

commit b68f9126 (2024-08-01): Show all com.victronenergy.battery services in Device List

gui-v2 need two separate battery lists, like this:

- In the Overview Battery drilldown, show all batteries from the list
in com.victronenergy.system/Batteries
- In the Device List, show all battery services under
com.victronenergy.battery.*

Fix Global.batteries.model to contain all battery services under
com.victronenergy.battery.*. The Overview Battery list can be created
specifically when requested, when the Battery widget is clicked.

Fixes regression from 5ebe75600bdf7ecacabc480a34fec153642355cf.

Fixes #1357
PR #1382 (2024-08-05): Show all com.victronenergy.battery services in Device List
gui-v2 need two separate battery lists, like this:

- In the Overview Battery drilldown, show all batteries from the list in com.victronenergy.system/Batteries
- In the Device List, show all battery services under com.victronenergy.battery.*

Fix Global.batteries.model to contain all battery services under com.victronenergy.battery.*. The Overview Battery list can be created specifically when requested, when the Battery widget is clicked.

Fixes regression from 5ebe75600bdf7ecacabc480a34fec153642355cf.

Fixes #1357
commit fa700f7c (2024-08-05): Show all com.victronenergy.battery services in Device List

gui-v2 need two separate battery lists, like this:

- In the Overview Battery drilldown, show all batteries from the list
in com.victronenergy.system/Batteries
- In the Device List, show all battery services under
com.victronenergy.battery.*

Fix Global.batteries.model to contain all battery services under
com.victronenergy.battery.*. The Overview Battery list can be created
specifically when requested, when the Battery widget is clicked.

Fixes regression from 5ebe75600bdf7ecacabc480a34fec153642355cf.

Fixes #1357
closed 2024-08-05 by blammit