victronenergy/gui-v2 #1360: Use full-publish-completed-echo in initial keep-alive

wiebeytec, 2024-07-25:
Starting Venus 3.50 (likely), dbus-flashmq has an additional feature in the keep-alive: `full-publish-completed-echo`. This is necessary to correctly request a topic list when using multiple concurrent clients, without creating race conditions.

A random token can be added to the initial keep-alive request:

```
{ "keepalive-options" : [ {"full-publish-completed-echo": "RANDOMSTRING" } ] }
```

And then you wait for the publication on `N/<portal ID>/full_publish_completed` with payload:

```
{"full-publish-completed-echo": "RANDOMSTRING", "value":1718860914}
```

See [this](https://github.com/victronenergy/dbus-flashmq?tab=readme-ov-file#keep-alive)

chriadam, 2024-07-25:
Is this something I should implement now, or do I need to wait for some prerequisite changes to dbus-flashmq to be integrated into venus?

wiebeytec, 2024-07-25:
It's included since Venus v3.50~5. I assume you don't have to worry about backwards compatibility or parallel version development? So that means you can just implement it.

chriadam, 2024-07-26:
https://github.com/victronenergy/gui-v2/pull/1375

chriadam, 2024-07-30:
Done.

closed 2024-07-30 by chriadam