victronenergy/gui-v2 #1361: Add critical loads to gui-v2

mpvader, 2024-07-25:
Nice to have for v1.0.0 - but non-blocking since wrt design not trivial.

Test systems (all with energy meter ofcourse):
- Multi RS three phase: [Groningen - RS3phase with EM](https://vrm.victronenergy.com/installation/394636/overview) - this is a non-critical test system in Groningen, update Venus OS when you need to.
- VE.Bus Multi three phase: 

The right side of the screen can now show up to three things:
- Loads
- EVCS (one widget for all of them.
- DC loads

So the challenge will be that this ia fourth box.

Wrt getting the data, that is easy, since its on D-Bus already.

Some options:
- add a fourth box, if that fits
- add it into the AC Loads box, and show both side by side, perhaps uses less space.

Note that more is coming, seeing heatpump for example. (cc @philipptrenz )

And also note that there is ongoing discussion wrt what the right wording is. Should we introduce "non-critical loads" as term? (I'm not convinced that we should), and what about essential loads, etc etc.

Screenshot:

[image]

jepefe, 2024-09-05:
Discussed with Serj, he is working on this.

blammit, 2024-09-09:
Serj has provided the design, this is ready for implementation.

blammit, 2024-09-22:
https://github.com/victronenergy/gui-v2/pull/1489

commit f955052e (2024-09-19): Overview: add Essential Loads widget

For the Overview page, the system can be configured to show either the
combined AC Loads, or separate Essential Loads and (non-essential) AC
Loads.

When the combined AC Loads is shown, this box shows the measurements
at com.victronenergy.system/Ac/Consumption.

When the loads are separated, the AC Loads box shows the measurements
at com.victronenergy.system/Ac/ConsumptionOnInput, and the Essential
Loads shows those at com.victronenergy.system/Ac/ConsumptionOnOutput.

The EVCS widget's connectors are now more complex, as it can connect
to Inverter/Charger, [... cut]
commit e580f714 (2024-09-19): Overview: add Essential Loads widget

For the Overview page, the system can be configured to show either the
combined AC Loads, or separate Essential Loads and (non-essential) AC
Loads.

When the combined AC Loads is shown, this box shows the measurements
at com.victronenergy.system/Ac/Consumption.

When the loads are separated, the AC Loads box shows the measurements
at com.victronenergy.system/Ac/ConsumptionOnInput, and the Essential
Loads shows those at com.victronenergy.system/Ac/ConsumptionOnOutput.

The EVCS widget's connectors are now more complex, as it can connect
to Inverter/Charger, [... cut]
commit 059d3f08 (2024-09-20): Overview: add Essential Loads widget

For the Overview page, the system can be configured to show either the
combined AC Loads, or separate Essential Loads and (non-essential) AC
Loads.

When the combined AC Loads is shown, this box shows the measurements
at com.victronenergy.system/Ac/Consumption.

When the loads are separated, the AC Loads box shows the measurements
at com.victronenergy.system/Ac/ConsumptionOnInput, and the Essential
Loads shows those at com.victronenergy.system/Ac/ConsumptionOnOutput.

The EVCS widget's connectors are now more complex, as it can connect
to Inverter/Charger, [... cut]
commit b7036b22 (2024-09-22): Overview: add Essential Loads widget

For the Overview page, the system can be configured to show either the
combined AC Loads, or separate Essential Loads and (non-essential) AC
Loads.

When the combined AC Loads is shown, this box shows the measurements
at com.victronenergy.system/Ac/Consumption.

When the loads are separated, the AC Loads box shows the measurements
at com.victronenergy.system/Ac/ConsumptionOnInput, and the Essential
Loads shows those at com.victronenergy.system/Ac/ConsumptionOnOutput.

The EVCS widget's connectors are now more complex, as it can connect
to Inverter/Charger, [... cut]
closed 2024-09-27 by blammit