victronenergy/gui-v2 #1368: GX (calibrated) temperature input precision

ReinvdZee, 2024-07-25:
As of recently, the temperature inputs of the Cerbo GX Mk2 are being calibrated. However, the temperature is shown with 0 decimals precision. In the device page, there is actually 1 decimal, but its always 0: 

[image]

If I heat the senser, it goes from 22.0 to 23.0 to 24.0, etc.

So we should decide on how many decimals we'd like to show for the GX temperature inputs and consistently implement that.
Since we made an effort to calibrate it, we should at least have 1 decimal precision. This would probably require a change to the backend somewhere, since I suspect the gui just pulls the value of dbus. 

@mpvader How many decimals should we have on the GX temp and GX tank 140 inputs? And should we distinguish between the calibrated and non-calibrated ones?

This will also be documented [here](https://docs.google.com/document/d/1yzHSpZoeZ0WOJJPhjivbhhD_iz37PaZtnZ9GpqVPh2M/edit#heading=h.82agtgjmtjv)

jhofstee, 2024-09-04:
Just remove the decimal to fix this. At least I can't think of any case were it is of any additional value.

chriadam, 2024-09-16:
Fixed.

closed 2024-09-16 by chriadam