victronenergy/gui-v2 #1373: 'Access Level' radio buttons allow selecting multiple access levels at the same time

DanielMcInnes, 2024-07-26:
I have 'Service' access level, and when I try to set the access level back to 'user' or 'installer', I can have multiple radio buttons selected at the same time
[image]

DanielMcInnes, 2024-07-26:
Yes it is. I didn't realise the behaviour was different when clicking the list item. For some reason, I always click the button part, rather than the list item part.

blammit, 2024-07-26:
Great, good to know. I'll verify this as part of those changes.

blammit, 2024-08-13:
https://github.com/victronenergy/gui-v2/pull/980 has been merged, verified this is okay now. Closing.

closed 2024-08-13 by blammit