victronenergy/gui-v2 #1377: Add support for DC gensets

martinbosma, 2024-07-26:
In VenusOS v3.50~5 we added support for DC gensets. This is missing for gui-v2.

The related gui-v1 changes are these:
- [Add dedicated page for gensets](https://github.com/victronenergy/gui/commit/b20574be6c3be329e5452928ee03e521c1c0e94d)
- [Add support for dcgenset service types](https://github.com/victronenergy/gui/commit/3763bb2ee1c5b34353b92705b30c3a3cd1a9919a)
- [PageGensetModel: add heatsink temperature (for Fischer Panda AGT)](https://github.com/victronenergy/gui/commit/b38cf07403bd7de69a5a0ff1a2d154487db679b9)
- [Add support for Hatz DC gensets](https://github.com/victronenergy/gui/commit/bb3bacdffb478baa6d774e2670b927fe657f28ad)
- [Add support for Fischer Panda AGT gensets](https://github.com/victronenergy/gui/commit/74e1ced96c018b48ef59c33dd1e8759dc47a43cb)

The following changes are related, but might already be included:
- [Genset: rename 'AutoStart' to 'RemoteStartModeEnabled'](https://github.com/victronenergy/gui/commit/f9c5cd6bc1e690f8573045b5efa17510f66b887d)
- [Refactored MbItemFpGensetError to MbItemGensetError](https://github.com/victronenergy/gui/commit/9030a2d9ba686d1e5eb57a89a9b39ef7e5ba043d)
- [PageGensetModel: use /Error/0/Id](https://github.com/victronenergy/gui/commit/2089e6ae3c828d1ffd4f59d18b18b013f70cb596)

martinbosma, 2024-07-26:
I will create recordings for venus-docker next week to test this. I'm not available for assistance from 5-26th of August.

martinbosma, 2024-08-02:
I've added simulation `gdf` for Fischer Panda AGT and `gdh` for Hatz gensets to venus-docker.

blammit, 2024-08-20:
> [PageGensetModel: use /Error/0/Id](https://github.com/victronenergy/gui/commit/2089e6ae3c828d1ffd4f59d18b18b013f70cb596)

Note: this will be addressed by https://github.com/victronenergy/gui-v2/issues/1268 so it can be ignored as part of this task.

commit ba95f975 (2024-09-03): Add support for DC gensets

Fixes #1377
commit 769539ab (2024-09-03): Add support for DC gensets

Fixes #1377
commit f54505e4 (2024-09-11): Add support for DC gensets

Fixes #1377
PR #1435 (2024-09-16): Add support for DC gensets
Fixes #1377
closed 2024-09-16 by DanielMcInnes