victronenergy/gui-v2 #1380: Grid unexpectedly shown as disconnected

blammit, 2024-07-31:
Tested on [this](https://vrm.victronenergy.com/installation/96310/dashboard) installation.

Overview grid looks like this:

[image]

Even though `mqtt/system/0/Ac/ActiveIn/Source = 1`:

[image]

And Brief page shows some Grid output:

[image]

blammit, 2024-07-31:
This is because `mqtt/system/Ac/In/0/Connected` is `true` instead of `1` when connected, on VenusOS v3.50~7. 

@ReinvdZee do you know if this is a deliberate change? On VenusOS v3.41~1 it is still `1` when connected, when querying the value on MQTT.

We can easily check for either `1` or `true` in gui-v2 to resolve this, but just want to check this change is intentional, and whether any other values have changed in a similar manner.

ReinvdZee, 2024-07-31:
@blammit I'm not sure if this is intentional or not. In dbus, it still has the value `1`. 

Maybe this was changed in the MQTT broker? 
@mpvader, @wiebeytec?

wiebeytec, 2024-07-31:
Dbus-flashmq looks at the dbus type for generating the json string. Dbus has a distinction between bool and int, so I'm curious what the underlying type is.

Can you run (something like) this on various Venus versions and post the output:?

```
dbus-send --print-reply=literal --system --dest=com.victronenergy.system /Ac/In/0/Connected com.victronenergy.BusItem.GetValue
```

Example result for an int:

```
dbus-send --print-reply=literal --system --dest=com.victronenergy.system /Relay/1/State com.victronenergy.BusItem.GetValue
   variant       int32 1
```

Example result for a bool:

```
dbus-send --print-reply=literal --system --dest=com.victronenergy.platform /ConnectVrmTunnel com.victronenergy.BusItem.GetValue
   variant       boolean true
```

ReinvdZee, 2024-07-31:
@wiebeytec 

On Venus OS 3.50~9, this is the output: 

```
root@einstein:~# dbus-send --print-reply=literal --system --dest=com.victronenergy.system /Ac/In/0/Connected com.victronenergy.BusItem.GetValue
   variant       boolean true
```

And on 3.41: 
```
dbus-send --print-reply=literal --system --dest=com.victronenergy.system /Ac/In/0/Connected com.victronenergy.BusItem.GetValue
```

In both cases, the values shows up in `dbus-spy` as `1`. So it seems the type has changed along the way.

wiebeytec, 2024-07-31:
Did you get no output from the second command, or did you make a copy/paste mistake?

ReinvdZee, 2024-07-31:
@wiebeytec,

Sorry, copy mistake. I updated the comment

wiebeytec, 2024-07-31:
I tried to see if i could find something in systemcalc or velib_python to explain this, but nothing obvious stands out.

I don't know at which end(s) it needs fixing, although it seems reasonable GUIv2 can handle both.

ReinvdZee, 2024-07-31:
I agree, Gui-v2 should be able to handle both. 

The path is indeed created by `dbus-systemcalc-py`. 
Maybe @izak has a clue?

izak, 2024-07-31:
My mistake guys...

It is the new feature where a system with no Inverter/Charger will show a connected grid connection if you have a grid meter. It is meant to better support systems with only a grid meter, and this is the line of code (that last `input_count==0` creates a bool instead of an int).

```
newvalues.update(self.input_tree(input_count, source.service, source.instance, t, input_count==0))
```

Easy fix, I will cast it to an int explicitly.

Introduced in v3.50~3. Affects only systems without an inverter/charger and with a grid meter.

blammit, 2024-08-01:
Thanks! I will also add a patch to allow gui-v2 to recognise either `1` or `true` as "connected".

commit 66ed44b2 (2024-08-05): Handle AC input /Connected value of either 1 or true

The backend may publish either 1 or true to indicate
/Ac/In/<x>/Connected is set, so allow for either value.

Fixes #1380
closed 2024-08-05 by blammit