victronenergy/gui-v2 #1384: Allow setting SSH password with new security profiles

chriadam, 2024-08-02:
Allow users to set SSH password, now that security profiles are supported.
Need to figure out precisely what is required for this task.

mpvader, 2024-08-02:
@jhofstee the plan is for that password, plus related settings to be settable on a dbus/mqtt  path/topic on com.victronenergy.platform; right?

Is that implemented already in venus-platform?

mpvader, 2024-08-02:
Correction: this was already added in the security settings branch; I didnt know that.

https://github.com/victronenergy/gui-v2/blob/4745a54018265358017bb502d901354011a683e8/pages/settings/PageSettingsGeneral.qml#L202

There is nothing to do here, (except for me adding it in the change log).  @chriadam

wiebeytec, 2024-08-02:
We need to re-open this, because it doesn't work. GUIv2 sets:

```
W/<portalid>/platform/0/Security/Api {"value":"{\"Action\":\"SetRootPassword\",\"Password\":\"w4mm9R8t00PPUFKF\"}"}
```

But it needs to do:

```
W/<portalid>/platform/0/Security/Api {"value":"{\"Action\":\"SetRootPassword\",\"SetRootPassword\":\"w4mm9R8t00PPUFKF\"}"}
```

jhofstee, 2024-08-02:
Untested, but this should fix it
```
                         //% "Password needs to be at least 8 characters long"
                         Global.showToastNotification(VenusOS.Notification_Info, qsTrId("settings_root_too_short_password"), 5000)
                     } else {
-                        var object = {Action: "SetRootPassword", Password: secondaryText}
+                        var object = {"SetRootPassword": secondaryText}
                         var json = JSON.stringify(object)
                         securityApi.setValue(json)
                     }
```

chriadam, 2024-08-05:
Slightly related - as per slack convo with Wiebe and Andrii, gui-v2 is also not sending the appropriate (write) message when changing to Secure profile.  We should fix that at the same time.

chriadam, 2024-08-08:
> Slightly related - as per slack convo with Wiebe and Andrii, gui-v2 is also not sending the appropriate (write) message when changing to Secure profile. We should fix that at the same time.

What's the best way for me to test this one?  At least, it seems to properly change to secure mode (and require password) when I change the setting now (with my PR above), both when connected to local nginx and when connected to VRM.

wiebeytec, 2024-08-08:
When I debugged it with Andrii, I only saw the MQTT side. It seems that password box in secure mode was the only problem; there was no problem in actually sending the `W` message.

If it works now, this ticket can be closed.

closed 2024-08-19 by chriadam