victronenergy/gui-v2 #1388: Generalize the code for opening pages based on service type

blammit, 2024-08-05:
There are a number of places where gui-v2 pushes device-setting pages based on service type. For example:

- Device List page
- Inverter/charger widget and InverterChargerListPage
- Battery widget and BatteryListPage
- Solar widget and SolarDeviceListPage

Ideally these would be consolidated into some globally-accessible function, e.g. `Global.openDeviceSettingsPage(serviceUid, properties)` that could push the appropriate page based on the service type extrapolated from the uid, and pass the specified properties when creating/pushing the page.

blammit, 2024-08-07:
The first step here is to ensure settings pages only require a serviceUid or bindPrefix, and do not require specific object types like `Battery` or `VeBusDevice`.

blammit, 2026-03-03:
This hasn't been resolved yet - we still hard-code the page urls for settings pages from various places.

DanielMcInnes, 2026-10-11:
The UI-test harness now works out where openDevicePage() can navigate by regex-parsing the return "..." lines in Global.devicePageUrl(), which ties product code to a fragile source-text rule and makes every caller look like it can open all 23 device pages. The route finder now sends every device page through the Inverter / Charger widget and the PageBattery target test fails.

To reproduce, from the build directory:

VENUS_GUI_TEST_CAPTURE_DIR=$(mktemp -d) ./bin/venus-gui-v2 --mock --mock-conf multi-rs --ui-test /pages/settings/devicelist/battery/PageBattery.qml

On main this ends 7 steps passed, 0 steps failed. On this PR it ends 1 steps failed, after Navigation step 1 opened '/pages/invertercharger/OverviewInverterChargerPage.qml' instead of expected '/pages/settings/devicelist/battery/PageBattery.qml'. The exit code is 0 either way.

blammit, 2026-10-12:
@DanielMcInnes did you mean to put that comment in the PR?

DanielMcInnes, 2026-10-12:
> 21\. [@DanielMcInnes](https://github.com/DanielMcInnes) did you mean to put that comment in the PR?

yes I did, is it ok?

chriadam, 2026-10-12:
Sure, but this is the issue, not the PR.
Note that the page-nav ui-test stuff is probably what I added previously, and not related to Toby's PR, per-se.

closed 2026-11-10 by blammit