victronenergy/gui-v2 #1389: Failing to build gui-v2 on macOS

itemir, 2024-08-05:
Following the instructions on [How to build venus gui v2](https://github.com/victronenergy/gui-v2/wiki/How-to-build-venus-gui-v2#patch-and-pack-the-webassembly-files), specifically *Building for Linux / Max OS X Desktop* on a macOS 14.5 but failing with the following error message:

```
itemir@Ilkers-MacBook-Pro build % ~/Qt/Tools/CMake/CMake.app/Contents/bin/cmake -DNO_CACHEGEN=true -DLOAD_QML_FROM_FILESYSTEM=true -DCMAKE_BUILD_TYPE=MinSizeRel ../
-- The CXX compiler identification is AppleClang 15.0.0.15000309
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at tests/CMakeLists.txt:6 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test HAVE_STDATOMIC
-- Performing Test HAVE_STDATOMIC - Success
-- Found WrapAtomic: TRUE
-- Found OpenGL: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.5.sdk/System/Library/Frameworks/OpenGL.framework
-- Found WrapOpenGL: TRUE
-- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR)
-- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR)
-- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR)
-- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR)
-- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR)
-- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR)
Building VenusOS for Darwin
CMAKE_CXX_FLAGS:
CMAKE_C_FLAGS:
Removing '-g' debug flag set by /opt/venus/dunfell-arm-cortexa7hf-neon-vfpv4/environment-setup-cortexa7hf-neon-vfpv4-ve-linux-gnueabi
new CMAKE_CXX_FLAGS:
Disabling cache generation for faster builds
This application will load qml sources from the filesystem, not from the compiled resources
-- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR)
-- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR)
components/controls/presseffect.frag -> components/controls/presseffect.frag.qsb exposed as ://components/controls/presseffect.frag.qsb
BUILD_DIR_TS_FILES: /Users/itemir/gui-v2/build/i18n/venus-gui-v2_ar.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_cs.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_da.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_de.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_es.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_fr.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_it.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_nl.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_pl.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_ro.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_ru.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_sv.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_th.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_tr.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_uk.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_zh_CN.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2.ts;/Users/itemir/gui-v2/build/i18n/venus-gui-v2_en.ts
CMake Error at CMakeLists.txt:1204 (add_dependencies):
  Cannot add target-level dependencies to non-existent target
  "venus-gui-v2_lrelease".

  The add_dependencies works for top-level logical targets created by the
  add_executable, add_library, or add_custom_target commands.  If you want to
  add file-level dependencies see the DEPENDS option of the add_custom_target
  and add_custom_command commands.

-- Found Python3: /usr/local/Frameworks/Python.framework/Versions/3.12/bin/python3.12 (found version "3.12.4") found components: Interpreter
CMake Warning at /usr/local/lib/cmake/Qt6LinguistTools/Qt6LinguistToolsMacros.cmake:399 (message):
  Failed to resolve language code for
  /Users/itemir/gui-v2/build/i18n/venus-gui-v2.ts.  Please update
  CFBundleLocalizations in your Info.plist manually.
Call Stack (most recent call first):
  /usr/local/lib/cmake/Qt6LinguistTools/Qt6LinguistToolsMacros.cmake:718 (_qt_internal_store_languages_from_ts_files_in_targets)
  /usr/local/lib/cmake/Qt6LinguistTools/Qt6LinguistToolsMacros.cmake:689:EVAL:1 (qt6_add_translations)
  CMakeLists.txt:DEFERRED

CMake Warning at /usr/local/lib/cmake/Qt6LinguistTools/Qt6LinguistToolsMacros.cmake:399 (message):
  Failed to resolve language code for
  /Users/itemir/gui-v2/build/i18n/venus-gui-v2_en.ts.  Please update
  CFBundleLocalizations in your Info.plist manually.
Call Stack (most recent call first):
  /usr/local/lib/cmake/Qt6LinguistTools/Qt6LinguistToolsMacros.cmake:718 (_qt_internal_store_languages_from_ts_files_in_targets)
  /usr/local/lib/cmake/Qt6LinguistTools/Qt6LinguistToolsMacros.cmake:689:EVAL:1 (qt6_add_translations)
  CMakeLists.txt:DEFERRED

-- Configuring incomplete, errors occurred!
itemir@Ilkers-MacBook-Pro build %
itemir@Ilkers-MacBook-Pro gui-v2 % pwd
/Users/itemir/gui-v2
itemir@Ilkers-MacBook-Pro gui-v2 %
```

Note that the path for qt-make provided in the instructions (`~/Qt6/6.5.2/macos/bin/qt-cmake`) is non-existent on a Mac, so replaced it with this: `~/Qt6/6.5.2/macos/bin/qt-cmake`

Qt 6.6.3 is installed and `qtmqtt` is built as described. 

```
itemir@Ilkers-MacBook-Pro gui-v2 % ls -l ~/Qt/
total 1560
drwxr-xr-x   5 itemir  staff     160 Mar 19 11:39 6.6.3
drwxr-xr-x   3 itemir  staff      96 Mar 19 11:53 Docs
drwxr-xr-x   3 itemir  staff      96 Mar 19 11:53 Examples
-rw-r--r--@  1 itemir  staff  397512 Aug  4 21:34 InstallationLog.txt
drwxr-xr-x@  7 itemir  staff     224 Aug  4 18:12 Licenses
drwxrwxrwx@  3 itemir  staff      96 Aug  4 18:12 MaintenanceTool.app
-rw-r--r--   1 itemir  staff  340263 Aug  4 21:34 MaintenanceTool.dat
-rw-r--r--   1 itemir  staff   25315 Aug  4 21:34 MaintenanceTool.ini
drwxr-xr-x@  3 itemir  staff      96 Jul 23 17:46 Qt Creator.app
lrwxr-xr-x   1 itemir  staff       3 Aug  4 21:49 Qt6 -> Qt6
drwxr-xr-x@  4 itemir  staff     128 Jun 20 05:46 QtDesignStudio
-rw-rw-r--@  1 itemir  staff    1592 Jun 20 05:46 README.txt
drwxr-xr-x@  5 itemir  staff     160 May 15 04:57 Tools
drwxr-xr-x  15 itemir  staff     480 Aug  4 21:54 build-qtmqtt
-rw-r--r--@  1 itemir  staff   10332 Aug  4 21:34 components.xml
drwxr-xr-x@  3 itemir  staff      96 May 24 03:22 dist
drwxr-xr-x@ 14 itemir  staff     448 Aug  4 21:33 installerResources
-rw-rw-rw-@  1 itemir  staff      27 Aug  4 21:34 licenseInfo.txt
-rw-r--r--@  1 itemir  staff     433 Aug  4 21:34 network.xml
drwxr-xr-x  16 itemir  staff     512 Aug  4 21:50 qtmqtt
itemir@Ilkers-MacBook-Pro gui-v2 % ls -l ~/Qt6
lrwxr-xr-x  1 itemir  staff  2 Aug  4 21:49 /Users/itemir/Qt6 -> Qt
itemir@Ilkers-MacBook-Pro gui-v2 %
```

Any advice will be appreciated. Thanks!

chriadam, 2024-08-05:
What version of cmake are you using?
You might be able to comment out line 1185 of the top level CMakeLists.txt and just make sure you build without parallelisation.

itemir, 2024-08-05:
Ok, commenting out this line (line 1204) helped and build now completes, thanks!

```
add_dependencies(${PROJECT_NAME}_lrelease ${PROJECT_NAME}_lupdate) # We shouldn't need this line, but without it, parallel builds (eg. 'make -j') do not generate translations correctly
```

Though the comment here is a bit spooky `without it, parallel builds (eg. 'make -j') do not generate translations correctly`. Anything of concern here?

DanielMcInnes, 2024-08-05:
> Ok, commenting out this line (line 1204) helped and build now completes, thanks!
>
> ```
> add_dependencies(${PROJECT_NAME}_lrelease ${PROJECT_NAME}_lupdate) # We shouldn't need this line, but without [...]

I found that parallel builds on a linux machine (eg. `make -j8`, or `cmake --build . -j8`) did something wrong with creating translations (i.e. the words and phrases shown on the screen of gui-v2). I can't remember exactly what it did wrong.

When you build the project, if you avoid parallel builds (eg. `cmake --build . ` instead of `cmake --build . -j8`) you shouldn't have any problems.

mr-manuel, 2024-08-05:
Did you check https://github.com/victronenergy/gui-v2/issues/492? Maybe it can help you :-)

mpvader, 2024-08-05:
Looks like you figured it out already, but the Qt6 version now in Venus OS, which also is the Qt6 version that all our devs working on gui-v2 use, is Qt 6.6.2.

itemir, 2024-08-06:
@mpvader , yes noticed it, that’s why I sticked with 6.6.2

itemir, 2024-08-06:
I think there is an opportunity to improve documentation and I might follow up with a PR later on but for now my problem is solved. Closing this ticket, thank you for the swift responses.

itemir, 2024-08-06:
Re-opening this as encountering a different problem. Happy to file a different issue but consolidating all build issues in a single issue looked more right. 

```
% ~/Qt/Tools/CMake/CMake.app/Contents/bin/cmake --build .
[ 54%] Built target VenusQMLModule_tooling
[ 54%] Generating themeobjects.h
Generating theme file /Users/itemir/gui-v2/src/themeobjects.h from theme directory /Users/itemir/gui-v2/themes/
[ 54%] Built target VenusQMLModule_autogen_timestamp_deps
[ 54%] Automatic MOC and UIC for target VenusQMLModule
[ 54%] Built target VenusQMLModule_autogen
[ 54%] Running AUTOMOC file extraction for target VenusQMLModule
[ 54%] Running moc --collect-json for target VenusQMLModule
[ 54%] Generating meta_types/qt6venusqmlmodule_minsizerel_metatypes.json
[ 54%] Automatic QML type registration for target VenusQMLModule
[ 54%] Generating .qsb/components/controls/presseffect.frag.qsb
spirv-opt -O /private/var/folders/tz/c3k9jy_579b6k3rjmx09rc6w0000gn/T/qsb-Yrlckx/qsb_spv_temp -o /private/var/folders/tz/c3k9jy_579b6k3rjmx09rc6w0000gn/T/qsb-Yrlckx/qsb_spv_temp_out
Failed to run spirv-opt -O /private/var/folders/tz/c3k9jy_579b6k3rjmx09rc6w0000gn/T/qsb-Yrlckx/qsb_spv_temp -o /private/var/folders/tz/c3k9jy_579b6k3rjmx09rc6w0000gn/T/qsb-Yrlckx/qsb_spv_temp_out: Child process set up failed: execve: No such file or directory
[ 54%] Running rcc for resource VenusQMLModule_raw_qml_0
[ 54%] Running rcc for resource qmake_Victron_VenusOS
[ 54%] Running rcc for resource venus-shaders
[ 55%] Building CXX object CMakeFiles/VenusQMLModule.dir/VenusQMLModule_autogen/mocs_compilation.cpp.o
[ 55%] Building CXX object CMakeFiles/VenusQMLModule.dir/venusqmlmodule_qmltyperegistrations.cpp.o
[ 55%] Building CXX object CMakeFiles/VenusQMLModule.dir/build/.rcc/qrc_qmake_Victron_VenusOS.cpp.o
[ 55%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/connman/connmandbustypes.cpp.o
[ 55%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/connman/cmmanager.cpp.o
[ 55%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/connman/cmmananger_interface.cpp.o
[ 55%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/connman/cmtechnology.cpp.o
[ 55%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/connman/cmtechnology_interface.cpp.o
[ 55%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/connman/cmservice_interface.cpp.o
[ 56%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/connman/cmservice.cpp.o
[ 56%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/connman/cmagent.cpp.o
[ 56%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/connman/cmagent_adaptor.cpp.o
[ 56%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/connman/clockproxy.cpp.o
[ 56%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/bms_error.cpp.o
[ 56%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/charger_error.cpp.o
[ 56%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/unit_conversion.cpp.o
[ 56%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/alternator_error.cpp.o
[ 56%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/ve_qitem.cpp.o
[ 57%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/ve_qitem_child_model.cpp.o
[ 57%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/ve_qitem_loader.cpp.o
[ 57%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/ve_qitem_sort_table_model.cpp.o
[ 57%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/ve_qitem_table_model.cpp.o
[ 57%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/ve_qitem_tree_model.cpp.o
[ 57%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/ve_quick_item.cpp.o
[ 57%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/ve_dbus_connection.cpp.o
[ 57%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/ve_qitems_dbus.cpp.o
[ 57%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/vebus_error.cpp.o
[ 58%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/veutil/src/qt/ve_qitems_mqtt.cpp.o
[ 58%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/aggregatedevicemodel.cpp.o
[ 58%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/basedevicemodel.cpp.o
[ 58%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/theme.cpp.o
[ 58%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/backendconnection.cpp.o
[ 58%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/language.cpp.o
[ 58%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/enums.cpp.o
[ 58%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/notificationsmodel.cpp.o
[ 60%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/clocktime.cpp.o
[ 60%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/cpuinfo.cpp.o
[ 60%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/frameratemodel.cpp.o
[ 60%] Building CXX object CMakeFiles/VenusQMLModule.dir/src/quantityinfo.cpp.o
In file included from /Users/itemir/gui-v2/src/quantityinfo.cpp:6:
In file included from /Users/itemir/gui-v2/src/quantityinfo.h:9:
In file included from /usr/local/lib/QtCore.framework/Headers/QObject:1:
/usr/local/lib/QtCore.framework/Headers/qobject.h:408:5: error: static assertion failed due to requirement 'std::is_pointer_v<Victron::Units::Units>': qobject_cast requires to cast towards a pointer type
    static_assert(std::is_pointer_v<T>,
[... cut ...]
/Users/itemir/gui-v2/src/quantityinfo.cpp:28:13: note: in instantiation of function template specialization 'qobject_cast<Victron::Units::Units>' requested here
        quantity = qobject_cast<Units>(Units::instance(nullptr, nullptr)).getDisplayTextWithHysteresis(unitType, value, quantity.scale, precision, unitMatchValue, formatHints);
                   ^
1 error generated.
make[2]: *** [CMakeFiles/VenusQMLModule.dir/src/quantityinfo.cpp.o] Error 1
make[1]: *** [CMakeFiles/VenusQMLModule.dir/all] Error 2
make: *** [all] Error 2
%
```

Suggestions?

itemir, 2024-08-06:
Ok, I think I figured it out. Few things:

- Above is caused by a conflict of Qt installation. There was one version installed via brew and another via Qt installer.
- Removing the brew installation brought additional problems. In oder to solve them: 
- You need to install Sharing Tools from Qt Maintenance Tool
- Set the following environment variable with something like: `export Qt6_DIR=/Users/itemir/Qt/6.6.3/macos/lib/cmake/Qt6/`

With above, could build a desktop version on macOS. All in all, macOS instructions could use some a bit more clarity.

itemir, 2024-08-06:
Closing with one last note for macOS. After a successful build, you can run it like this:

```
open bin/venus-gui-v2.app --args --mock
```

This is slightly different than Linux as you pass the parameters with `--args` argument to macOS open.

closed 2024-08-06 by itemir

chriadam, 2024-08-08:
Thanks, also updated the wiki a bit.