victronenergy/gui-v2 #1391: Changing VRM device instances does not work

ReinvdZee, 2024-08-05:
On Venus OS v3.50~13. Changing VRM device instances does not work as expected. 
I can change the VRM instance number and a reboot dialog shows up when leaving the menu. However, after rebooting, the VRM instance has not changed. I tried changing the SmartShunt VRM ID from 279 to 280, but the ID remains 279 after the reboot.

[image]

And it also lists a lot of unconnected devices. I don't think it's supposed to do that? 
I also noticed that the reboot dialog shows after editing a field and then editing it back to the original value. In this case, no ID's have changed, so returning to main menu should not invoke the reboot dialog.
[image]

@blammit, @chriadam @DanielMcInnes
Can you take a look? Its a bit urgent.
You can use [this system](https://vrm.victronenergy.com/installation/323213/overview) to test which has a smartshunt connected.

ReinvdZee, 2024-08-05:
Setting the deviceinstance manually with dbus-spy works: 
[image]
After the reboot, the smartshunt also shows up in VRM with the new deviceinstance.

blammit, 2024-08-06:
Sure, will look into this.

blammit, 2024-08-06:
@ReinvdZee I have a tentative fix for this, and will have a PR ready tomorrow. Regarding this point:

> And it also lists a lot of unconnected devices. I don't think it's supposed to do that?

The list shows all items under `com.victronenergy.settings/Settings/Devices` that have a valid `com.victronenergy.settings/Settings/Devices/ClassAndVrmInstance` value. If no /CustomName or /ProductName is available then it shows "Unconnected". This was discussed when the issue was created in https://github.com/victronenergy/gui-v2/issues/305 but perhaps there's a better way to do this now? It would certainly be nicer to find a better display name.

ReinvdZee, 2024-08-06:
@blammit 

Good that you have a fix already. Regarding the unconnected devices being displayed, this is not urgent and I don't really have a better way for now.

blammit, 2024-08-07:
https://github.com/victronenergy/gui-v2/pull/1395 should resolve this issue.

I found some sorting issues in the list while testing; have filed https://github.com/victronenergy/gui-v2/issues/1396 for that.

commit 70e416f4 (2024-08-09): Change the VRM instance for the right model index

Whe setting the VRM instance, use the uid and deviceClass to locate
the model data to be changed. Do not use the model index, as the
classAndVrmInstanceObjects model may differ from the
classAndVrmInstanceModel model that is used for display (e.g. if the
list order has changed) so the model index is not reliable.

Fixes #1391
closed 2024-08-09 by blammit