victronenergy/gui-v2 #1404: Problems with showAccessLevel and writeAccessLevel

mr-manuel, 2024-08-08:
Change the access level to `User`.

## ListSwitch
The user can modify the switch, but it should be read-only by setting `writeAccessLevel`.
```qt
ListSwitch {
    id: switchTailscaleEnabled
    //% "Enable Tailscale"
    text: qsTrId("settings_tailscale_enable")
    dataItem.uid: Global.systemSettings.serviceUid + "/Settings/Services/Tailscale/Enabled"
    writeAccessLevel: VenusOS.User_AccessType_Installer
    enabled: isRunning
}
```
The `showAccessLevel` is working correctly.

## ListTextField
The user can modify the text box, but it should be read-only by setting `writeAccessLevel`.
```qt
ListTextField {
    //% "Hostname"
    text: qsTrId("settings_tailscale_hostname")
    dataItem.uid: Global.systemSettings.serviceUid + "/Settings/Services/Tailscale/Hostname"
    placeholderText: "--"
    enabled: !isEnabled
    writeAccessLevel: VenusOS.User_AccessType_Installer
}
```
The `showAccessLevel` is working correctly.

## ListButton
The button is shown, but it should not be shown by setting `showAccessLevel`.
```qt
ListButton {
    //% "Logout from Tailscale account"
    text: qsTrId("settings_tailscale_logout_from_tailscale_account")
    //% "Logout"
    button.text: qsTrId("settings_tailscale_logout")
    allowed: isConnected
    showAccessLevel: VenusOS.User_AccessType_Installer
    onClicked: commandItem.setValue ('logout')
}
```
The `writeAccessLevel` is hiding the element. It seems that `showAccessLevel` and `writeAccessLevel` are swapped.

mr-manuel, 2024-08-08:
Can be tested with https://github.com/victronenergy/gui-v2/pull/1393

blammit, 2024-08-09:
@mr-manuel  The problem is that when you set `enabled`, it overwrites the binding that controls whether the control is read-only. So when you set `enabled`, include `userHasWriteAccess` in the binding, like this:

```
ListSwitch {
    writeAccessLevel: VenusOS.User_AccessType_Installer
    enabled: isRunning && userHasWriteAccess
}
```

Similarly, when you set `allowed`, it overwrites the binding that controls whether the button is visible. Do this instead:

```
ListButton {
    allowed: defaultAllowed && isConnected
    showAccessLevel: VenusOS.User_AccessType_Installer
}
```

I can see this is confusing, and ideally, it wouldn't be possible to overwrite these `enabled` and `allowed` so easily and lose the default behavior. But, this is the API we have for now, and it can be improved later on.

blammit, 2024-08-09:
As discussed, this is not a bug, just an issue with how the enabled/allowed state can be overwritten in the API. Closing.

closed 2024-08-09 by blammit

mr-manuel, 2024-08-09:
I was just documenting this, but you were faster. Thanks.