victronenergy/gui-v2 #1405: Generator drilldown on WASM doesn't work

ReinvdZee, 2024-08-08:
On Venus OS 3.50~14. 
The WASM correctly shows the blue outline on the generator widget:
[image]

But the drilldown does not work. It opens an empty page. Locally, this works properly. 
This is on a system with a vebus system connected, if that matters. This can be tested with the DSE simulation on venus-docker.

@blammit
I suppose we need to publish the instance number of the main generator? 
As indicated by @izak's comment [here](https://github.com/victronenergy/vrm-portal/issues/314#issuecomment-1980630339) (issue needs to be reopened when this is fixed) this may also be useful for VRM.

blammit, 2024-08-09:
Hm, I'm not sure what's going on here. I tried connecting to [this](https://vrm.victronenergy.com/installation/366384/overview) installation, which does have a DSE genset, and I was able to open the generator drilldown there.

> I suppose we need to publish the instance number of the main generator?

The drilldown page shows `PageAcIn.qml` for the genset that is being shown, so it shouldn't be necessary to get an instance number separately.

ReinvdZee, 2024-08-13:
@blammit,

I'm testing on [this](https://vrm.victronenergy.com/installation/402872/dashboard) system. It works properly with the DSE genset simulator of venus-docker. 

However, when I connect the system to the physical genset controller, it behaves as I described in my first comment. Small demonstration:

[Screencast from 13-08-24 13:45:06.webm](https://github.com/user-attachments/assets/9bebd7ea-aaf9-4c2a-915c-005a4af76c7a)

If I open gui-beta on VRM, it works properly, though. The GX is running Venus os 3.50~14.

You can test it on the system. Be careful not to start the generator, its connected to a physical genset controller.

blammit, 2024-08-27:
Note: discussed with @ReinvdZee that this can't be reproduced now. Will have to check again further down the track.

mpvader, 2024-10-30:
Is this bug still reproducible @ReinvdZee

ReinvdZee, 2024-10-30:
@mpvader,
Tested on the [ESS system](https://vrm.victronenergy.com/installation/473156/dashboard) with 3.50~42.

There actually is no drilldown anymore:
[image]

The generator is there and AC input 1 is configured as "Generator".
Same behavior is on [this](https://vrm.victronenergy.com/installation/366384/overview) system (see @blammit's [comment](https://github.com/victronenergy/gui-v2/issues/1405#issuecomment-2277192139))

blammit, 2024-10-31:
Thanks, will look into this.

PR #1663 (2024-11-01): Overview: allow drilldown on non-active AC input
Any AC input widgets on the Overview page should be clickable and take the user to the drilldown/settings page, even if it is not the currently active AC input.

Also allow PageAcIn.qml and PageRsSystem.qml to provide the device name internally, instead of relying on the launching page to provide the name via the 'title' property.

Fixes #1405
commit ff7fba3c (2024-11-01): Overview: allow drilldown on non-active AC input

Any AC input widgets on the Overview page should be clickable and take
the user to the drilldown/settings page, even if it is not the
currently active AC input.

Also allow PageAcIn.qml and PageRsSystem.qml to provide the device
name internally, instead of relying on the launching page to provide
the name via the 'title' property.

Fixes #1405
closed 2024-11-01 by blammit