victronenergy/gui-v2 #1408: Changing user role to 'User' or 'User & Installer' doesn't work anymore

ReinvdZee, 2024-08-12:
Tested on Venus OS v3.50~14. When attempting to set the access level to 'User' or 'User & Installer', the password input box disappear and the desired user level is not selected: 

[Screencast from 12-08-24 14:55:01.webm](https://github.com/user-attachments/assets/79f576f7-bda1-43d8-9e9f-925f7403f7b9)

This happens when transitioning from any access level to 'User' or 'User & Installer'.

mr-manuel, 2024-08-12:
Should be fixed with https://github.com/victronenergy/gui-v2/pull/1402

In the meanwhile you can still enter the password and hit enter.

ReinvdZee, 2024-08-12:
Oh alright, already fixed. 
Thanks, I'm closing this one.

blammit, 2024-08-12:
Not yet fixed -- I'll reopen and mention this in the commit in #1402, then this can be closed automatically when that PR is merged.

commit 5bbe22d3 (2024-08-12): Only show the password field for the selected radio button

Set the allowed binding according to whether the password field is
for the currently selected option, and do not overwrite the binding.

Fixes #1408
commit 1d4d0f5f (2024-08-12): Only show the password field for the selected radio button

Set the allowed binding according to whether the password field is
for the currently selected option, and do not overwrite the binding.

Fixes #1408
commit 2af55d8a (2024-08-13): Only show the password field for the selected radio button

Set the allowed binding according to whether the password field is
for the currently selected option, and do not overwrite the binding.

Fixes #1408
closed 2024-08-13 by blammit