victronenergy/gui-v2 #1409: Genset page: service & runtime metrics

ReinvdZee, 2024-08-13:
Some work is done related to run time and service metrics (see [issue](https://github.com/victronenergy/venus/issues/1315)). Following changes apply to the connected genset menu, as well as the relay-controlled (generator start/stop) menu, unless otherwise noted.  

- [ ] Remove “Operating time” of connected gensets from the “Engine” page.
- [ ] Move the “Run time and service” from settings to the top-level and add some items. This menu should contain:
a. Total run time. This field is synchronized with `/Engine/OperatingHours` (if valid). If the path does not exist (relay controlled generator) or is invalid (unsupported by connected genset), the total run time entry should be modifiable like it is now.
b. Daily run time counters (submenu)
c. Daily run time counters reset button
d. Time to next test run (if enabled)
e. Service timer (if nonzero service interval)
f. Service interval setting
g. Service timer reset button
- [ ] Rename “Run time” to “Current run time” and show it on the top-level if the generator is running, warming up or cooling down. (`/State` equal to 1, 2 or 3).
- [ ] Reorder the menu structure of the generator start/stop (relay) to be the same as that of the genset menu.

The new run time & service menu:
[image]
[image]

ReinvdZee, 2024-08-19:
The new "Run time and service" menu looks like this:
[image]

I also moved the "Accumulated running time since last test run" entry here. It is only visible when test runs are enabled and the user has access level service.

mpvader, 2024-09-05:
discussed today in the meeting @blammit will look into this herself, with help from Philipp, if she gets to this one before Rein is back.

blammit, 2024-09-09:
@ReinvdZee I've made a PR at https://github.com/victronenergy/gui-v2/pull/1448 using the patch you sent me earlier. I added some checks for empty UIDs, as we discussed previously.

commit 212d0ceb (2024-09-09): Genset UI: Run time and service changes

- Remove Operating time from the Engine page.
- Sync total run time with genset `/Engine/OperatingHours` (dbus_generator)
- Add 'Run time and service' menu at top level
- Remove Total run time from top level since its part of the run time and service menu.
- Change 'Run time' to 'Current run time' on top-level
- Only show 'Current run time' when generator is running, warming up or cooling down.
- Move service interval setting to run time and service menu
- Reordered menu items of generator start stop (relay).
- Changed toast message when disabling the s [... cut]
PR #1448 (2024-09-09): Genset UI: Run time and service changes
- Remove Operating time from the Engine page.
- Sync total run time with genset `/Engine/OperatingHours` (dbus_generator)
- Add 'Run time and service' menu at top level
- Remove Total run time from top level since its part of the run time and service menu.
- Change 'Run time' to 'Current run time' on top-level
- Only show 'Current run time' when generator is running, warming up or cooling down.
- Move service interval setting to run time and service menu
- Reordered menu items of generator start stop (relay).
- Changed toast message when disabling the se [... cut]
commit b86d5a99 (2024-09-09): Genset UI: Run time and service changes

- Remove Operating time from the Engine page.
- Sync total run time with genset `/Engine/OperatingHours` (dbus_generator)
- Add 'Run time and service' menu at top level
- Remove Total run time from top level since its part of the run time and service menu.
- Change 'Run time' to 'Current run time' on top-level
- Only show 'Current run time' when generator is running, warming up or cooling down.
- Move service interval setting to run time and service menu
- Reordered menu items of generator start stop (relay).
- Changed toast message when disabling the s [... cut]
closed 2024-09-09 by blammit