victronenergy/gui-v2 #1432: Improve Generator card display of running state and runtime

blammit, 2024-08-30:
The current generator card looks like this:

[image]

There are two issues here:

- The "Running by SOC condition" text is longer than what's desired in the design
- The runtime should be clear about whether it's showing hours-minutes or minutes-seconds. When under a minute, it's clear that it's seconds because the number will change every second, but the UI should be more explicit about this.

Updated design from Serj is like this:

[image]
[image]

- It should show the generator state and condition, separated by a dot if necessary. If it was auto-started (i.e. any start condition other than "Manual start" or "Test run") then show "Auto-started" as the state text.
- Show a "s" suffix when the runtime is less than a minute, otherwise show the "hh:mm" format without a suffix.

*Note*: The timer shown in the generator start/stop dialog (shown when "manual stop" is pressed) should also be updated to show the runtime in the same format.

DanielMcInnes, 2024-10-08:
@blammit do you have a link to Serj's updated design for this?

blammit, 2024-10-09:
@DanielMcInnes that wasn't specified, but I believe the text should be "Manually started" in that case.

commit e7a35c64 (2024-10-09): Improve Generator card display of running state and runtime

Fixes #1432
commit 66b5959b (2024-10-16): Improve Generator card display of running state and runtime

Fixes #1432
commit 6c2d14d6 (2024-10-21): Improve Generator card display of running state and runtime

Fixes #1432
commit 013d74cc (2024-10-23): Improve Generator card display of running state and runtime

Fixes #1432
closed 2024-10-23 by DanielMcInnes